JMRI/JMRI #9084: Splitval not working correctly when only parts of the low CV is used

KPolmann, 2020-10-17:
I have a problem in JMRI version 4.20+Rc7ba8249b on Java version 1.8.0_261 (en).
I am working with a decoder, where some of the LocoNet Addresses are spread over 2 CVs.
The low CV, bits 4-7 are containing the lower part of the LocoNet address and the whole CV+1 (all 8 bits) is containing the higher part of the LocoNet address.
See code examples in the attachment. 

If I try to press “Write full sheet” from the tab in question, the actual CVs in the decoder are not updated correctly and the background in the input fields turns in red color.
If I instead go to the tab CVs and press “Write full sheet”, then the actual CVs in the decoder is updated correctly.

When inspecting the contents of the CVs in the tab CV (after only writing the LocoNet address in the LocoNet address field on  the tab in question), the content is correct.

If I try to read the LocoNet address in the field of the tab in question by pressing “Read full sheet” then the field also turns red.

When using the code (of course with the right masks) on LocoNet addresses where the whole lower CVs (all 8 bits) is used for the lower part and bits 0-3 in CV+1 is used for the higher part of the LocoNet address, it work correctly.
 
Does anyone have a solution to this problem?
[The following are examples of different layouts of the addresses in the CVs.docx](https://github.com/JMRI/JMRI/files/5395199/The.following.are.examples.of.different.layouts.of.the.addresses.in.the.CVs.docx)

dheap, 2020-10-17:
@KPolmann Please checkJMRI version 4.21.2. There have been some changes since V4.20 that _may_ have fixed this.

Also, there have been some longstanding issues with overlapping read/writes when using “Read full sheet” and “Write full sheet”. these have not been tracked down.

Please supply a simple actual decoder XML file that demonstrates the issue and that we can check rather than what appears at first sight to be pseudo-code (or maybe XSLT, which I'm not familiar with).

We need to start with a simple actual case we can test.

KPolmann, 2020-10-17:
Unfortunately 4.21.2 shows exactly the same behaviour.

[MGP_Panel.zip](https://github.com/JMRI/JMRI/files/5395669/MGP_Panel.zip)

silverailscolo, 2020-10-17:
As noted by @dheap 4.21.12 supports a new style to specify different masks for each CV, and you try that on your case. Instead of the familiar pattern:
```
 <variable CV="2" item="Short Address" mask="VVVVXXXX" default="1" minOut="1">
        <splitVal highCV="4" upperMask="VVVVVVVV"/>
        <label>Accessory Address</label>
    </variable>
```
you can now use syntax:
```
 <variable CV="2,4,8" item="AccAddr" mask = "VVVVXXXX VVVVVVVV" default="1" minOut="1">
        <splitVal/>
        <label>Accessory Address</label>
    </variable>
```
Note the 3rd CV. Use spaces instead of commas to separate the masks.
When more CV's than masks are supplied, the last mask will also be applied to all higher CV,s (CV8 in this example).

dheap, 2020-10-17:
@KPolmann, @silverailscolo  I only saw this issue posted at bedtime last night so didn't have time to look into it properly. I realised later that it's an issue I encountered with the ESU SwitchPilot Servo V2.0 definitions. Take a look at the warning on the Basic pane of that definition.

The problem is not with the masks per se, but with the fact that the `Order n Type` enumVal shares the same CV as the `Order n Address` splitVal (albeit different bits). During Read/Write Sheet operations the splitVal and enumVal options overlap, the propertyChangeListeners get intermingled and _very bad things happen_ with the entire Read/Write operation, affecting subsequent Read/Write values.

In my SwitchPilot testing, I was able to "prevent" the problem by moving the enumVal to a different sheet, but that's not a solution.

At the time I looked at the code, but couldn't see where the problem lay.

I'll post more as I get time today...

dheap, 2020-10-18:
I've done some tests of "Read All Sheets" versus "Read All CVs" with an ESU "SwitchPilot Servo V2.0 and attached log files from my NCE Power Cab as well as CV dumps, etc.

- A "Compare Full Sheet" shows that the  "Read All CVs" has returned the correct values, but that the "Read All Sheets" has messed up most of the CV reads.
- Note that in the NCE log for "Read All Sheets" there is a duplicated read of CV35 (the one that appears in both a splitVal and an enumVal on the Basic pane, but using non-overlapping bitmasks. All CV values read after that are showing correct values in the NCE log, but all subsequent reads have the wrong value assigned in the roster entry.
- In almost all cases this represents an offset of 1 between the CV number and assigned value. The few exceptions can be explained by min/max/mask effects.
- Since propertyChangeListeners get only a Value and not the associated CV number, I have failed to pin the problem down.
- I can avoid the problem by moving the splitVal and enumVal to different sheets, so it seems the splitVal and enumVal processing are interacting and causing the problem.

@KPolmann This doesn't solve your problem, but is recorded here to help track down the problem.

[SwitchPilot Read Sheets Problem.zip](https://github.com/JMRI/JMRI/files/5396991/SwitchPilot.Read.Sheets.Problem.zip)

dheap, 2020-10-20:
@KPolmann I've spent quite a bit of time investigating the splitVal code to see if I can find the problem, but failed.

However, I did discover a simple workaround with my "SwitchPilot Servo V2.0" definition. simply by swapping the order of the splitVal and the enumVal in the pane from:
```
            <display item="Second Address Enable" format="checkbox"/>
            <display item="Second Address" tooltip="Sets base address for Secondary"/>
```
to:
```
            <display item="Second Address" tooltip="Sets base address for Secondary"/>
            <display item="Second Address Enable" format="checkbox"/>
```
caused the problem to go away.

@KPolmann Could you please try the same? For example, change from:
```
		<display item="Combo {$index} Order Type 1">
			<label>Order 1 Type</label>
		</display>
		<display item="Combo {$index} Order Address 1">
			<label>Order 1 Address</label>
		</display>
		<display item="Combo {$index} Order Type 2">
			<label>Order 2 Type</label>
		</display>
		<display item="Combo {$index} Order Address 2">
			<label>Order 2 Address</label>
		</display>
```
to:
```
		<display item="Combo {$index} Order Address 1">
			<label>Order 1 Address</label>
		</display>
		<display item="Combo {$index} Order Type 1">
			<label>Order 1 Type</label>
		</display>
		<display item="Combo {$index} Order Address 2">
			<label>Order 2 Address</label>
		</display>
		<display item="Combo {$index} Order Type 2">
			<label>Order 2 Type</label>
		</display>
```
I know it's not a long-term solution, but would be helpful if you tried this change with your MGP Panel decoder and reported back whether it resolved the problems with that particular pane.

[ESU_SwitchPilot swapped.zip](https://github.com/JMRI/JMRI/files/5406268/ESU_SwitchPilot.swapped.zip)

KPolmann, 2020-10-20:
The proposal mentioned by dheap works without any problems. It never occurred to me to switch the order of the display item.
Thanks a lot for the support.

dheap, 2020-10-20:
@KPolmann I'll leave this open as the underlying issue needs resolving eventually.

dheap, 2021-03-30:
This is an ongoing problem but I have not been able to find the cause.

dheap, 2021-05-14:
This is still an ongoing problem.

bobjacobsen, 2023-03-29:
I believe that the underlying issue was fixed with the SplitEnumVariableValue fix(es) in #11786 and #11790

closed 2023-03-29 by bobjacobsen