JMRI/JMRI #9095: Suggested Tooltips documentation improvements.

dheap, 2020-10-22:
On a [jmriusers](https://groups.io/g/jmriusers/message/181461) thread, user Mike Heintzman has proposed some useful improvements to the documentation:

JMRI: Applications>JMRI Features and Tools:(last section)

**Throughout the various screens in the JMRI applications you will find setting specific Help information called Tooltips. These tips, which are pop-ups accessed by hovering your mouse over the setting, can give you information specific to that setting, such as what CV controls a decoder setting in the Comprehensive Programmer, how to add and set-up a turnout in PanelPro, or getting the most out of Operations. You can turn Tooltips on in DecoderPro by going to Edit>Preferences>Roster>Programmer(tab) and click on ‘Show CV numbers in tool tips.’ In PanelPro panel design, you can set when to turn on Tooltips by opening your panel and go to Options>Show Icon tooltips. The length of time Tooltips are displayed is controlled in Edit>Preferences>Display>'Tool tip display time' in both DecoderPro and PanelPro. The Default is set to 4 seconds. It’s suggested to set it to 20 seconds if you use them.**

JMRI Setup>Setting your Preferences>Display:

-  **How long tooltips (where available) should be shown. Tooltips are setting specific help information. They are accessed by hovering your mouse over the setting.**

bobjacobsen, 2020-10-22:
I think the "Show CV numbers in tool tips" option is about, well, showing the CV numbers in the tool tips vs just showing the text that was defined in the decoder file.  The tool tips are always on, in the sense that anything in the file will be displayed, but it sometimes doesn't look like that because lots of decoders don't carry `<tooltip>` elements or attributes.  Here's an example:
```
      <variable CV="10" default="0" item="EMF Droop Config" tooltip="Range 0-252">
        <decVal max="252"/>
        <label>EMF Feedback Cutoff Speed Step</label>
        <label xml:lang="it">Soglia sensibilità EMF</label>
      </variable>
```

and another

```
            <variable CV="15" item="Decoder Lock ID" default="0">
                <decVal/>
                <label>Decoder Lock ID</label>
                <label xml:lang="de">Decoder Lock</label>
                <label xml:lang="ca">Bloqueig decoder per ID</label>
                <tooltip>Set this to match value below and decoder will unlock</tooltip>
                <tooltip xml:lang="de">Setzen Sie diesen Wert auf den unten stehenden Wert und der Decoder wird entsperrt</tooltip>
                <tooltip xml:lang="nl">Stel hier dezelfde waarde in als ID hieronder en de decoder kan geprogrammeerd worden</tooltip>
            </variable>
```

dheap, 2020-10-22:
Point taken. But I think that, with some modifications, this can be made into a useful contribution.

bobjacobsen, 2020-10-22:
Completely agree!

MikeHeintzman, 2020-10-22:
I'd thought about that yesterday, as it happens, and added another line:
Throughout the various screens in the JMRI applications you will find setting specific Help information called Tooltips. These tips, which are pop-ups accessed by hovering your mouse over the setting, can give you information specific to that setting, such as what CV controls a decoder setting in the Comprehensive Programmer, how to add and set-up a turnout in PanelPro, or getting the most out of Operations. You can turn Tooltips on in DecoderPro by going to Edit>Preferences>Roster>Programmer(tab) and click on ‘Show CV numbers in tool tips.’ In PanelPro panel design, you can set when to turn on Tooltips by opening your panel and go to Options>Show Icon tooltips. The length of time Tooltips are displayed is controlled in Edit>Preferences>Display>Tool tip display time in both DecoderPro and PanelPro. The Default is set to 4 seconds. It’s suggested to set it to 20 seconds if you use them.
Not all settings have Tooltips. Some settings are self explanatory, and others are too complicated for a short hint. In those cases see the listing in the JMRI Help files.

icklesteve, 2020-10-23:
I think there's a strong case for upping the initial install ToolTip display time.

Whenever I do a first-time profile I up to 30-40 secs, we're dealing with some potentially complicated concepts throughout.
If the tooltip is in the way, I can just move the mouse.
Rather than adding another step for new users, perhaps just up the default to 20s?
Experienced users will more likely be aware of where the time can be adjusted, but the fewer hurdles in place for new users, the better.

icklesteve, 2020-10-24:
So I checked for Tooltip display time guidelines.

MS defaults to 5secs. ( https://docs.microsoft.com/en-us/dotnet/api/system.windows.forms.tooltip.autopopdelay )
Apple reccomend 10 secs ( via https://balsamiq.com/learn/resources/ui-control-guidelines/tooltips/ )

I could put some code in to up JMRI to default 10secs if people think worthwhile?

MikeHeintzman, 2020-10-24:
I'd agree with Steve, but I think you need to experiment some to see what effect, if any, displaying the tooltip blocks the user from entering data into the setting. Is the tooltip over the top of the box blocking data entry? That may be why the default is so low.

bobjacobsen, 2020-10-24:
Each Look&Feel should be providing its own default values.  That’s the idea behind the L&F mechanism: it provides a consistent look and feel. Some are arbitrary-but-consistent. Others are specific to a platform. 

It’s possible to that JMRI is already overriding this. It might be good to check for that and remove it if found. 

I’d prefer that the L&F-provided defaults not be changed, though I’m fine if somebody wants to add a preference or two to change user-specific values.

MikeHeintzman, 2020-10-24:
I'd be fine setting the default to 10 seconds or just leaving it at 4. I think the object here is to educate users as to what and where Tooltips are, and where they can modify them.

icklesteve, 2020-10-25:
> It’s possible to that JMRI is already overriding this. It might be good to check for that and remove it if found.

I had a look into this the other day, no JMRI defaults found.

> I’d prefer that the L&F-provided defaults not be changed, though I’m fine if somebody wants to add a preference or two to change user-specific values.

No probs, I'll leave as is, there's already a preference for the display time, which is more than most applications!
Thanks for the guidance Bob.

MikeHeintzman, 2020-10-28:
Based on the comments above I had a go at rewriting it. Here goes:

Throughout the various screens in the JMRI applications you will find setting specific Help information called Tooltips. These tips, which are pop-ups accessed by hovering your mouse over the setting, can give you information specific to that setting. For example, In Roster>Labels & Media>Roster Media, hovering over the image block advises you how to add an image to your Roster entry. In PanelPro, Edit mode, hovering over a track entry type will tell you how to add a Turnout, Track. etc. In Operations>Trains hovering over the buttons at the bottom of the screen tells you what those buttons are for. 
Not all settings have Tooltips. Some settings are self-explanatory, and others are too complicated for a short hint. In those cases, or for further information, see the listing in the JMRI Help files.
The length of time Tooltips are displayed in both DecoderPro and PanelPro is controlled in Edit>Preferences>Display>Tool tip display time. The Default is set to 4 seconds. It’s suggested that new users set the display time to 20 seconds. The time can be reduced or increased according to your preference.
In addition, Tooltips can be modified within applications. In your Roster, Program Mode, you can have the CV associated with a setting displayed (or not displayed) by going to Edit>Preferences>Roster>Programmer(tab) and click on ‘Show CV numbers in tool tips.’ In PanelPro, you can set when to display Icon Tooltips by opening your panel and go to Options>Show Icon tooltips and setting it to your preference. 

One of the things I struggled with was where to put this. I suggested Features and Tools because that seemed to be a good fit, but I'm half convinced that it should have it's own entry in the Help list. You've been looking at this longer than I have, and I'm sure you have a good idea where it should go. That's just my zwei pfennig.

bobjacobsen, 2020-10-28:
That looks very helpful.  Just two points:

"The Default is set to 4 seconds" - Is this true?  It shouldn't be. For example, the default on the Nimbus L&F, at least for macOS, should be 10 seconds. That's how it's standardly defined by Swing for all Nimbus-using applications.  It might be that we're overriding the L&S default, in which case we should fix what's doing that in JMRI.

"It’s suggested that new users set the display time to 20 seconds" - not sure what the basis for that is.  Can we make it a bit less definitive?  It's generally not a good idea to leave new users with the idea that they have to go through setting optional options; they tend to mess things up. Maybe "You can set this longer, perhaps to 20 seconds, if you want more time to read them and don't mind them hanging around"

As to having its own item in help:  Tooltips are a really standard thing.  They're not just a JMRI thing. Google "tooltip" and you'll get over 39 _million_ hits.  So I think people mostly need to find it when they're looking for it. I think it would be reasonable to add some of the text to expand the tool tip discussion here:

https://www.jmri.org/help/en/package/apps/TabbedPreferences.shtml#Display

which can be linked from the 25 or so references to tool tip in the existing help, plus in the Table of Contents and Index for the help files, and perhaps in the [Glossary](https://www.jmri.org/help/en/glossary/index.shtml) (which often doesn't get enough love).

MikeHeintzman, 2020-10-28:
I just created a new profile and the time was set to 4 seconds. I'm on a Windows 10 PC if that makes any difference.

I like something like your verbiage better, although it brings up the whole Holy Hand Grenade of Antioch discussion.

With the number of msgs on the forum stating they have no computer skills I thought it might be best to explain what a Tooltip was and make sure they know they're available in JMRI. Just my thoughts.

And I forgot to make reference to putting it in the Glossary. I looked at the #Display and it came down to choosing there or Tools and Features. I chose poorly.

dheap, 2020-10-28:
> @bobjacobsen Maybe "You can set this longer, perhaps to 20 seconds, if you want more time to read them and don't mind them hanging around"

You can always dismiss them earlier by moving the mouse away slightly.

MikeHeintzman, 2020-10-29:
I thought one of the possible problems might be the Tooltip field partially blocking the entry field until time ran out. However I just did a quick survey  in different apps and found that the Tooltips were enough out of the way to not interfere with data entry. They also disappear when you click the data entry box, so by clicking on the entry box or by moving the mouse the Tooltip disappears.  So either way I don't think setting the default time to 10 seconds is detrimental, if it can be done easily. If it can't I'd just leave it at 4.

closed 2021-04-14 by stale[bot]