JMRI/JMRI #9102: LocoNet simulator gives no data into LocoNet Monitor.

positron96, 2020-10-25:
Given that Simulator can be (and is) used for troubleshooting the LocoNet protocol and data being sent, it would be very useful for simulator to show what data is being send/received.

Would it be possible to enable echoing messages from simulator to monitor?

bobjacobsen, 2020-10-26:
I just tried this, and messages sent by JMRI seemed to show in the monitor. I tried the power and turnout controls.

Can you be more specific on where you're seeing messages not sent?

Note that the simulator isn't an all-items simulator:  It has no way to actually simulate a layout.  So it doesn't i.e. send feedback messages from turnouts.

positron96, 2020-10-26:
Hi, tanks for a response. I've just tried with turnouts, and they so work for me as well.
What does not show in the Monitor is CV Programming data. It seems to be simulated (when reading, I get successful reading of value 123 on all CVs), but there are no packets in the monitor.

bobjacobsen, 2020-10-26:
Right.  There's no (simulated) command station out there to handle the communications, so the LocoNet simulator uses a programming simulation that doesn't use the LocoNet connection.  The CVs start at 123, but should remember any value that's written to each CV.

The best way to see how specific real hardware talks is to hook up to it.  There are lots of little differences between the various generations of command stations.

positron96, 2020-10-27:
Ah, I see. Well, I'm developing a firmware for that real hardware as a DIY project, so I can't really expect it to communicate in a reasonable manner until I program it to.
Anyway, thank you for the information!
I'm going to close the issue for now, since it seems that the effort to implement the feature is way bigger than the benefits it provides.

closed 2020-10-27 by positron96