JMRI/JMRI #9139: Programmer won't open

mousefixer, 2020-11-09:
I am using JMRI 4.20 on a windows 10 system through a Digitrax DS52.
When I open JMRI I am able to correctly read the identity of the locomotive on my programming track, but when I click on the "Program" button, nothing happens. When I right click on the locomotive in the roster and click program, I get the same result. 
At the same time, a JMRI throttle won't control a loco on the layout, and turnout and power contol also do not work. Basically the only thing that is working fine is reading the ID of a loco on the programming track.
Any suggestions?

silverailscolo, 2021-02-20:
Can you add the actual computer-to-layout connection in use? The [DS52](https://www.digitrax.com/products/stationary-decoders/ds52/) is a turnout decoder, so that won't ID anything. 
If your issue has been solved since you posted, please close this.
Egbert

mousefixer, 2021-02-20:
Hi there. Thanks for responding. If I wrote DS52, that was an error- I
meant DCS52- the "Zephyr Plus". The JMRI program opens on my computer, (PC
with windows 10) and my roster appears. I know the computer is talking to
the DCS52, because I can identify a loco on the programming track from
JMRI. But then when I click on "Program" nothing happens. I'm pretty sure
the problem is within JMRI itself. I have tried uninstalling and
reinstalling it, but the problem persists. I welcome any suggestions you
may have. Thanks.
Ron Bremer

silverailscolo, 2021-02-21:
Can you describe your actual computer-to-DCC connection, what connects your USB port to LocoNet?

mousefixer, 2021-02-22:
Hi again. My Digitrax DCS52 "Zephyr Express" command station is connected
directly to my PC via USB cable, and my programming track is connected to
the DCS52.
Thanks for asking- I'll take any help I can get at this point!
Ron Bremer

silverailscolo, 2021-02-28:
Ron, I see now how your connection is set up (I use the first Zephyr without the USB, but I checked your model is available in JMRI). Your LocoNet is working OK from the Zephyr? (Rails and the programming track don't rely on LocoNet).
When you open the LocoNet Monitor from the LocoNet menu, do you see any messages when you run a train using the Zephyr throttle? That would confirm the hardware connection working both ways.
You sure have read the win install and FAQ help?
Please open the JMRI System  Console (from the Help menu), copy all text (from a simple start up session plus your decoder I'd step) and share it here.
Egbert

mousefixer, 2021-02-28:
Hello again. Thank you so much for your help. I hadn't thought to use the
loconet monitor as a way of seeing what was going on, Using that as a
starting point I was able to troubleshoot and fix the problem. Now
the programmer is working and I am back in business.
Thanks and have a great day!
Ron Bremer

mousefixer, 2021-02-28:
OK I may have spoken too soon. I was able to add a new locomotive to my
roster, read all of its CVs from the programming track, program its loco
ID, etc.. However when I place a loco that is already in my roster on the
programming track the programmer still won't open. I am able to identify
the locomotive correctly from the roster, but when I click on "Program"
nothing happens. So I know for sure that there is two-way communication
between JMRI on my PC and the loco on my programming track, but for some
reason JMRI won't launch the programmer.

KenC57, 2021-02-28:
Two possible things come to mind. First, check the console for odd error messages. That's via Help->System
Console, copy/paste what you find into an email (it might be many screens) if you see something you don't
understand. But if you do find something, leave at least a couple of lines before and after what you don't
understand to give us the context of what's happening. If in doubt, copy a lot more than not enough.

Next is it did open, but not on the screen where you can see it. Look in the Windows menu item and see if it
shows a programmer or roster entry open. If so, that says it thinks it is there, you just can't see it for
some reason. Depending on which computer OS you are using, there are different keystrokes to find errant
windows.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.org
www.syracusemodelrr.org

mousefixer, 2021-02-28:
OK thanks again- I am looking into those things now.

mousefixer, 2021-02-28:
One thing I just noticed is that on my hard drive under
c:/users/Ron/JMRI/Ron/Programmers- that "programmers" file folder is empty.
Should it be?

mousefixer, 2021-02-28:
This is the error log I found, and it seems to be saying that there are
missing files in the /programmers/ file folder.

KenC57, 2021-02-28:
Since you have a C:\ in the file path, I figure you are Windows based. The programmers directory in the JMRI
user data is frequently empty. I am slightly confused as the path has your username in it twice.
Normal: c:/users/Ron/JMRI/Programmers
Yours: c:/users/Ron/JMRI/Ron/Programmers

This also presumes the application is in the normal location of C:/Program Files (x86)/JMRI
-ken c

dsand47, 2021-02-28:
Ken,

Maybe his profile name is "Ron".

Dave Sand

dheap, 2021-03-01:
@mousefixer I can't see your log, but this problem is not uncommon. For some unknown reason the filename of the currently-selected programmer format gets corrupted (two .xml extensions at the end). 

Go to Preferences->Roster->Programmer and change the Format to something other than the current setting (it most likely will be blank). Save and Restart JMRI.

When JMRI restarts, change it back to your preferred format. I prefer "Advanced" but others prefer "Comprehensive". Save and restart JMRI.

Now try opening your roster entry.

Dave

mousefixer, 2021-03-02:
Hi. Thanks for weighing in on this issue. I was seeing some instances of
the double .xml extensions. I think this is getting me closer to rooting
out the problem. Thanks.

dheap, 2021-03-02:
@mousefixer
>Hi. Thanks for weighing in on this issue. I was seeing some instances of the double .xml extensions. I think this is getting me closer to rooting out the problem. Thanks.

The only place I've seen it before is in "<program folder>/xml/programmers" and the log reported that it couldn't find
"<program folder>/xml/programmers/(Basic|Advanced|etc.).xml.xml"

In these cases it was simply a case of corrupted preference files and a change/save fixed it.

The problem appeared when trying to open any roster entry and prevented you from doing so.

Dave

KenC57, 2021-03-02:
The double .xml.xml on the programmer type is easily fixed by selecting a different programmer
(Edit->Preferences->Roster) and then save/exit. Reselect the one you want when you reopen JMRI. Then you can
save/restart and it should be ok.

closed 2021-05-01 by stale[bot]