JMRI/JMRI #9143: MQTT issues in 4.21.2

jerryg2003, 2020-11-10:
Trying to learn about MQTT (and also planning on updating documentation for new features).  Running mosquitto broker locally (localhost).  Found the following issues:

Set MQTT channel to "jgtrains/"
Setting up MT1001, MS1001, ML1001, MR1001 (kept all "topics" at default):

(1) Can't set up MR1001.  Only "internal" listed on Reporter table pulldown so can't set up a Reporter under MQTT.
(2) Set up ML1001 OK.
(3) Setting up MT1001 with sensor MS1001, got a pop-up window warning and this in the console:
2020-11-10 16:50:56,708 beantable.TurnoutTableAction          WARN  - Requested Turnout MT1001 uses same address as Light ML1001 [AWT-EventQueue-0]
(4) then tried to add one-sensor feedback, found my sensor in the pulldown, but got these messages in the console:
2020-11-10 16:52:28,357 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - MT1001 [AWT-EventQueue-0]
2020-11-10 16:52:28,357 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - MT1001 [AWT-EventQueue-0]
(5) However, the TO was set up OK with sensor attached.
(6) Then I closed the mosquitto broker, got this in the console:
2020-11-10 17:10:04,557 mqtt.MqttAdapter                      ERROR - Unable to reconnect [MQTT Rec: ri9CB6D09A38D53db5210dM]
Unable to connect to server (32103) - java.net.ConnectException: Connection refused: connect
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:75)
	at org.eclipse.paho.client.mqttv3.internal.ClientComms$ConnectBG.run(ClientComms.java:538)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.net.ConnectException: Connection refused: connect
	at java.net.DualStackPlainSocketImpl.waitForConnect(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(Unknown Source)
	at java.net.AbstractPlainSocketImpl.doConnect(Unknown Source)
	at java.net.AbstractPlainSocketImpl.connectToAddress(Unknown Source)
	at java.net.AbstractPlainSocketImpl.connect(Unknown Source)
	at java.net.PlainSocketImpl.connect(Unknown Source)
	at java.net.SocksSocketImpl.connect(Unknown Source)
	at java.net.Socket.connect(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:66)
	... 2 more

If I'm doing something wrong, please let me know.  Also, happy to do other testing if that would be useful.

Jerry

bobjacobsen, 2020-11-10:
The code for MQTT Reporters doesn’t exist yet.

jerryg2003, 2020-11-11:
Regarding MQTT Reporters:  I was actually led down that path by your comments on my post in the MQTT Forum and the fact that the MQTT connection page in JMRI lists "Reporter topic" along with turnout, sensor, and light topics.  Obviously whoever wrote that page got ahead of themselves.  I will patiently await their arrival...

Other issues still seem valid though?

jerryg2003, 2021-03-15:
Keeping issue up-to-date:  confirmed in 4.23.2 that scenario in lines 3-5 still exists.  Java exception thrown for line 6 still occurs.

closed 2021-05-14 by stale[bot]