JMRI/JMRI #9145: Track width differences between Layout Editor panel and web client

mstevetodd, 2020-11-11:
There are some inconsistencies with how track widths are drawn in panel.js. Screenshot and example panel provided by Robin Becker. I verified this with the current Jenkins build.
[track width drawing demo.xml.txt](https://github.com/JMRI/JMRI/files/5520969/track.width.drawing.demo.xml.txt)
[image]
[image]

geowar1, 2020-11-12:
I suspect I know what the issue is but won't be able to take a look within the next two weeks.
Unless @dsand47 wants to take a shot at it. (hint: PR#4433.)

geowar1, 2020-11-13:
Just committed a fix in LayoutEditor:misc bug fixes PR #9129.

geowar1, 2020-11-19:
If you can verify this as fixed by PR #9129 then please close… Thx.

closed 2020-11-19 by geowar1

mstevetodd, 2020-11-21:
Verified in latest Jenkins build. Thanks!