JMRI/JMRI #9147: Java exception adding non-existent sensor to Panel

jerryg2003, 2020-11-12:
Using JMRI 4.21.2.

Opened Panel Editor to create a new Panel.  Clicked on "Sensor" in the drop down for "Select the type of icon to Add to Panel:"  Since no sensors were yet defined, I typed "is001" in the "Add item to table" entry box.  Got this exception:

2020-11-12 09:38:25,146 managers.AbstractManager              ERROR - Invalid system name for Sensor: "CSis001" must be an integer after "CS". [AWT-EventQueue-0]
2020-11-12 09:38:25,146 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
jmri.NamedBean$BadSystemNameException: "CSis001" must be an integer after "CS".
	at jmri.jmrix.cmri.CMRISystemConnectionMemo.validateSystemNameFormat(CMRISystemConnectionMemo.java:441)
	at jmri.jmrix.cmri.serial.SerialSensorManager.validateSystemNameFormat(SerialSensorManager.java:267)
	at jmri.Manager.makeSystemName(Manager.java:165)
	at jmri.managers.AbstractManager.makeSystemName(AbstractManager.java:101)
	at jmri.Manager.makeSystemName(Manager.java:136)
	at jmri.Manager.makeSystemName(Manager.java:114)
	at jmri.managers.AbstractProvidingProxyManager.provideNamedBean(AbstractProvidingProxyManager.java:50)
	at jmri.managers.ProxySensorManager.provideSensor(ProxySensorManager.java:45)
	at jmri.jmrit.picker.PickListModel$SensorPickModel.addBean(PickListModel.java:571)
	at jmri.jmrit.picker.PickListModel$SensorPickModel.addBean(PickListModel.java:1)
	at jmri.jmrit.display.IconAdder.addToTable(IconAdder.java:602)
	at jmri.jmrit.display.IconAdder.lambda$0(IconAdder.java:525)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

Continued on without closing JMRI.  Changed "is001" to "001" and PE added sensor CS1 to the sensor table and put an icon on the panel.

[I note that on a previous invocation of PanelPro, I had added CMRI sensors to another panel file.  However, in this case, no panel file was yet loaded.  Curious as to how one would have added an internal sensor to a panel using the "Add item to table" entry box?]

Jerry

dsand47, 2020-11-12:
Jerry,

If you had typed IS001 it would create IS001.  If you type 001 it would create CS1.

The lower case "is" resulted in an attempt to add the default CS prefix to is001 which is an error since CSis001 is not a valid CMRI system name.

The first connection provides a default prefix for creating system names when the entered value does not have a valid prefix.

Dave Sand

icklesteve, 2020-11-12:
As the prefix for Internal Sensor ( IS ) did not match the prefix you entered ( is ), the default Sensor Manager ( CMRI ) was used.

Perhaps a quick fix would be to update the check in the AbstractProvidingProxyManager to case insensitive?
( capitalise both strings before comparison ).
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/managers/AbstractProxyManager.java#L229

Further down the process, In makeSystemName, the actual System Name should return the system prefix capitalised, leaving any lower case text intact.

Would also be good add a try / catch to IconAdder
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/display/IconAdder.java#L598

until someone gets a chance to add pre-click "Create" validation check to the UI.

Steve.

jerryg2003, 2020-11-12:
Thanks all.  I'll add this "hint" to my list to update documentation.  However, I don't think the user should get a Java exception for making a typing mistake.  I would agree that lower case characters should be interpreted as upper for the purposes of entering system names.  Seems like a common UX practice.

Jerry

dsand47, 2020-11-12:
Jerry,

Only the prefix has to upper case.  The rest of the system name depends on the connection and bean type requirements.  For example, a virtual signal mast:  IF$vsm:MilwHD:SL-D($0007).

Dave Sand

KenC57, 2020-11-12:
The create tool splits selecting the prefix from the rest of the system
name. It sounds like the first time you had the CMRI connection selected in
the connection pull-down. Your entry of is001 should be kicked as an invalid
name, I though we finally were giving user friendly messages about invalid
values?

The correct way to enter Internal elements would be to choose the 'Internal'
from the connector pull-down and then your identifier for which internal
thing you were creating. In your case, if you had selected the 'Internal'
connection first, it would have created the sensor "ISis001" as the system
name.

This is the pattern, split the connection name from the rest of the system
name for enter to insure valid names are created.

dsand47, 2020-11-12:
Ken,

Jerry was using the Panel Editor => Edit => Add Sensor which allows you to create sensor table entries on the fly.  

Dave Sand

bobjacobsen, 2020-11-12:
I agree that that a raw exception isn't the way to handle this.  That `jmri.NamedBean$BadSystemNameException` should be caught in the JMRI GUI code and presented to the user in some way. Then the user can type the right string, instead of JMRI trying to guess.

But it's not a good idea to try to guess what the user meant, i.e. by "interpret as upper case" 

It's perfectly possible to have `i` or `i2` or lots of other lower case system prefixes.  There are people who use those to i.e. distinguish `M` and `m` for two different CAN-based connections.  

The sensor type letter `S` is always upper case, but how would JMRI do "interpret as upper case" for '`is`? Is it 'iS` unless that doesn't exist, in which case its `IS`?  What if it's a typing error?

We should _NOT_ edit the user input string, i.e take lower case as upper case.  People have put huge amounts of time into doing that, but every effort failed. We always end up with "well, it worked here, and it didn't work there, it worked different over there, and it got this case wrong". As a group, we're not organized enough to do it consistently and permanently.  

So we decided to not do that. Several times. Including pulling out code that did it partially.

jerryg2003, 2020-11-12:
Bob - 

Thanks for the explanation and reasoning on upper vs lower case.  I didn’t realize that iS and IS could both be appropriate system prefixes for a sensor, although as I re-read  https://www.jmri.org/help/en/html/doc/Technical/Names.shtml, I see that it is so stated (in my defense, if one is needed, all the examples on this page only use upper case, as is the list of default names - a documentation oversight I am about to fix).  

As to my issue, I hope we can all agree that users shouldn’t end up with a Java exception sitting in the system console (which I suspect many don’t regularly monitor) rather than some pop up saying something helpful, no matter what they enter into JMRI.

Thanks again,
Jerry

jerryg2003, 2021-03-15:
Keeping this issue up-to-date, confirmed that exception still thrown in 4.23.2 (note that help pages have been updated to warn users about sensor naming, use of upper and lower case, etc.).

PR #9582 (2021-03-25): IconAdder - Fix Create New Bean Exception
Add Dialogue warning on create Bean Exception.
Add Bean / Connection specific ToolTip to address input field
Add Bean type to Add Button
Improve constructor
Fixes #9147 although longer term would be good to add prior validation.

[image]
closed 2021-03-25 by danielb987