JMRI/JMRI #9149: Warrants - Signal Table - Offset column - zero value problem

sidlo64, 2020-11-12:
Warrants - Occupancy block - Signal Table
Problem with editing Offset column
Non-zero number is without problem.
Zero value cannot be changed to other value. 
When you try to change the value, zero is returned.

[image]

sidlo64, 2020-11-14:
Circuit Builder - Add/Edit Signal Masts
Edit SignalMasts
Field Entrance space adjustment 
After saving the panel, the value zero will always be saved
[image]

silverailscolo, 2020-11-14:
A fix for the Warrants - Occupancy block - Signal Table - Offset entry issue is included in PR #9142, coming up. Will ave it reviewed to see if my correction is, well, correct.

In the panel I see a different issue (wasn't working on that): it doees save some value, but teh value is about 10x smaller when I reopen. WIll check how this is handled.

sidlo64, 2020-11-15:
The displayed values are in cm (/ inch), but the stored values are in mm - 10 times higher.

sidlo64, 2020-11-15:
First tests

The Edit Signal window was too small when first opened. I had to enlarge it. Only after the third restart of PanelPro did the window remember its size.

[image]

Each time you open the Edit Signal window, the units are set to inches.

[image]

Rows in OBlock tables are too low. Compare with the Signals Masts table. There is a normal height.

[image]

You cannot enter a negative number in the field Offset. This is reset when focus leaving the field.

[image]

silverailscolo, 2020-11-16:
Thanks for looking this. Initial size of the signal edit pane and the negative entry have been addressed. (hard to simulate a "fresh" use after working on it for some time).
Table row heigth is a surpise, will check code but don't see that here:
[image]

sidlo64, 2020-11-16:
Table row height

OBlockTablePanel.java
line 354
table.setRowHeight((new JButton(**"X"**).getPreferredSize().height)*9/10);

TableFrames.java
line 63
private static final int ROW_HEIGHT = (new JButton(**"X"**).getPreferredSize().height)*9/10;

Window Add New Turnout has probably bad layout

[image]

Edit Signal
Notice.
The table row is from TomZac**Trat** to TomZac**Zahl**
It is flipped in the dialog box from TomZac**Zahl** to TomZac**Trat**
If you don't notice, you unintentionally swap blocks.

[image]

The Edit Signal window has a small height when first opened.

[image]

silverailscolo, 2020-11-20:
Thanks. I increased the initial height of the pane. Don't think the Oblocks are presented in the wrong order at the start, but I did correct the right hand Block label to Protected Block. PR was merged, feel free to add your own improvements🙂

closed 2020-12-08 by silverailscolo