JMRI/JMRI #9165: BossBlockLogic error from keeping SSL when all sensor, signals, turnouts removed from SSL

KenC57, 2020-11-20:
This leaves errors like:
2020-11-20 11:58:37,794 automat.Siglet                        ERROR - Siglet start invoked for "3EWBlockBossLogic", but no inputs provided [AWT-EventQueue-0]

For the test case, use LocoNet Simulator.
XoverTest.xml is the before file.
I then went to SSL and selected 3EW.
I cleared the sensor Blk3E and signal XCC entries and hit apply.
The bossblock error showed in the console.
I saved the file as XoverTest3.xml
When loading XoverTest3.xml, you get the error in the console.
What was left in the file is:
    <signalelement signal="3EW" mode="1" limitspeed1="false" limitspeed2="false" useflashyellow="false" distantsignal="false">
      <comment />
    </signalelement>
Instead of it being deleted. Only fix is manual edit of XML or removal of signal head. Bob J says it should be removed when the file is written.
[XoverTest.zip](https://github.com/JMRI/JMRI/files/5575032/XoverTest.zip)
Used zip file since xml isn't accepted file format and you'd want to see before and after anyway.

danielb987, 2020-11-20:
> What was left in the file is:

If you enter text that has `<something>` it will not be displayed unless you use `&lt;something&gt;`

But if you select the text and then click on the `<>` icon (insert code), the text will be treated as code and displayed correctly.

dsand47, 2021-03-30:
As of JMRI 4.23.3 the SSL dialog includes a Delete button.

closed 2021-03-30 by dsand47