JMRI/JMRI #9169: Not starting from exe

Flyinhhigh, 2020-11-22:
I'm new to DCC,NEW Install  downloaded "JMRI.4.20+Rc7ba8249b" Installed 
Ran : LaunchJMRI.exe and received this error message:-"

"No command line parameter.UsageJMRI.exe [/debug] [/noisy] [/profileID.] [JOPTION]
[--cp:a=CLASSPATH} [--cp:p=CLASSPATH] class {config] 

Win10 pro , RAM 16gb , i7 8700K , 6core, 3.70GHz , CPU Water Cooler , ROG, Z370-E Motherboard, GTX 108 0 Ti ,11 GB Video Card

Java Ver:  jre-8u271-windows-x64 (lastest)

I have no Dcc layout as I am researching and wanted to see how JMRI worked.

Thanks

Chris.

dheap, 2020-11-22:
Chris,

> I'm new to DCC,NEW Install downloaded "JMRI.4.20+Rc7ba8249b" Installed
> Ran : LaunchJMRI.exe and received this error message:-"

You can't just run LaunchJMRI.exe and expect it to work.

If you did the default installation, the JMRI installer should have installed some shortcut icons (DecoderPro, PanelPro) on the desktop and in your Start Menu.

These shortcuts have all the required command line parameters for LaunchJMRI.exe.

Dave in Australia

Flyinhhigh, 2020-11-22:
Thanks. I'll try that.

closed 2021-02-20 by silverailscolo