JMRI/JMRI #9170: Latest release doesn't recognise Doehler & Haass SD05A sound decoder

BobLeicester, 2020-11-22:


Re465, 2020-11-24:
It looks to me that it's. a new decoder.
This decoder has the same manual as the SD10A.
This guide is from June this year.

I don't have time to write a decoder definition.
I hope Klb4ever (Ronald Kuhn) reads along, because he wrote the previous definition.

Currently, you can use SD10A 1.11+. 

Re 465

BobLeicester, 2020-11-24:
Hi,

Thank you for the reply, I will use the SD10A definition then.

Regards
Bob

From: Re465
Sent: 24 November 2020 08:37
To: JMRI/JMRI
Cc: BobLeicester; Author
Subject: Re: [JMRI/JMRI] Latest release doesn't recognise Doehler & HaassSD05A sound decoder (#9170)

It looks to me that it's. a new decoder.
This decoder has the same manual as the SD10A.
This guide is from June this year.
I don't have time to write a decoder definition.
I hope Klb4ever (Ronald Kuhn) reads along, because he wrote the previous definition.
Currently, you can use SD10A 1.11+.
Re 465
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2020-12-16 by danielb987