JMRI/JMRI #9171: DS64 config tool missing "ButtonCancel" resource

devel-bobm, 2020-11-23:
Lack of this resource causes an exception when the "Reset DS64 to Factory Defaults" button is pressed, and no LocoNet traffic is generated.

danielb987, 2020-11-24:
"ButtonCancel" is defined at row 325 in java/src/jmri/NamedBeanBundle.properties which is the root bundle.
jmri.jmrix.Bundle extends jmri.Bundle.
jmri.jmrix.loconet.Bundle extends jmri.jmrix.Bundle.
jmri.jmrix.loconet.ds64.Bundle extends jmri.jmrix.loconet.Bundle.

When I try this dialog, I get the error that the resource `ButtonTextResetToFactoryDefault` is missing.

In the bundle, there is a string `ButtonResetToFactoryDefault`. Should this name be used instead? See `java/src/jmri/jmrix/loconet/ds64/Ds64.properties`.

devel-bobm, 2020-11-25:
When I initially reproduced the problem, I inserted the "ButtonCancel" resource.  Then on re-run, I encountered the "ButtonTextResetToFactoryDefault".  There seem to be other missing resources.

I was not able to track down the problem as I seem to have a "corrupted" local git repository.  

I have not been able to spend time resolving those problems. I hope to be able to follow-up in the next few days.

devel-bobm, 2020-11-26:
The inheritance chain from `jmri.jmrit.loconet.ds64.Bundle` reaches up to `jmri.Bundle`, but, for some reason, `ButtonCancel` does not get recognized.

I've simply added the missing resource at the `jmri.jmrix.loconet.DS64` level.  And I fixed a couple other issues I found along the way.  See #9178.

devel-bobm, 2020-11-27:
P/R #9178 was canceled, in favor of new P/R #9181 .

PR #9181 (2020-11-28): Fix DS64 configuration tool ability to perform factory reset
- Resolves #9171 by fixing inconsistent I18N keys and restoring a missing key.
- Fixes problem where resetting to factory defaults _ignored_ the selected Board Address and reset any DS64 board configured as Board Address 1.

This P/R replaces canceled P/R #9178.
closed 2020-11-28 by danielb987