JMRI/JMRI #9177: MQTT Client Authentication

mattgyver83, 2020-11-26:
Currently it doesn't look like MQTT support for JMRI does not support supplying usn/pw credentials for authentication to the broker.  I'm leveraging mosquitto so I was able to setup the broker configuration to allow anonymous access to the /trains/ topic on a specific port which only JMRI can access however I would really prefer to be able to instead provide credentials in the MQTT configuration so I can not require a specific broker config for JMRI and the /trains/ topic.  Could this MQTT usn/pw authentication be considered for a future release?

mattgyver83, 2021-01-05:
While I'm not a java developer I have taken to task implementing this and am nearing a reliable state I just need to do a few more sanity checks.  Once I have this complete I'll contribute the code back upstream and hopefully someone more proficient than I can assist in optimizing it so it can be contributed as a future feature.  Hoping to have a commit ready by the end of the week or sooner.

mattgyver83, 2021-01-07:
Submit PR #9278

monkeyslayer56, 2021-03-15:
Also interested in this.

danielb987, 2021-05-15:
A new PR, #9794, extends #9278, to fix some issues with that PR. So once #9794 is merged, this issue is expected to be resolved.

closed 2021-06-30 by bobjacobsen