JMRI/JMRI #9179: Edit Logix window - Add new Conditiontionals - will not be displayed

sidlo64, 2020-11-27:
In the Edit Logix window after adding a new conditionals, it will not appear in the list of conditionals. After closing the window and reopening the Edit Logix window, the new Conditionals is visible.

dsand47, 2020-11-28:
Petr,

I am not able to re-create your symptoms.  What version of JMRI are you running?  Are there any message in the System Console (Help => System Console)?

Dave Sand

sidlo64, 2020-11-28:
Today Jenkins JMRI.4.21.3ish+jenkins+20201128T0517Z+R024723f4a3.tgz 
Tested on Linux and Windows. I can do it on MacOS too when necessary.

Tested on Linux. Connection Digitrax - Loconet Simulator.

How to reproduce:

Open Logix. Conditional1 is in dhe list. Add new Conditional2.

[image]

Click Update. In Logix list is still only Conditional1.

[image]

Close Edit Logix window. Open Edit Logix window. Conditional2 is in list.

[image]

System Log is empty.

dsand47, 2020-11-28:
The behavior change for adding conditionals in the List Editor appears to be the caused by PR #9047.

dsand47, 2020-12-07:
The latest development build has the fix for this issue.

closed 2020-12-07 by sidlo64