JMRI/JMRI #9183: Zimo decoder models are not detected accurately

chrisridd, 2020-11-28:
The New loco wizard correctly detects the Zimo decoder in my Fleischmann 706483 loco (decoder fitted by Fleischmann). I can see it reading CV 8 (145 = Zimo), CV 7 (37 = firmware version) and then CV 250 (177 = Zimo decoder model)

However it highlights all versions and models of all Zimo decoders, which suggests that it is not interpreting CV 7 or 250 correctly in the Zimo case.

It would be helpful to correctly detect the Zimo decoder models and versions.

Zimo document at http://www.zimo.at/web2010/documents/MX-KleineDecoder.pdf how to interpret CV 250 (see page 5) and it looks like a value of 177 means Zimo MX617.

dheap, 2020-11-29:
@chrisridd Thanks for reporting this. I'm looking into it.

dheap, 2020-11-29:
@Klb4ever I've done some edits and am running some checks on a PR to address this. I'd like your review once I've raised it.

dheap, 2020-11-29:
@chrisridd @Klb4ever See #9187 for a resolution.

dheap, 2021-03-30:
Resolved by #9187

closed 2021-03-30 by dheap