JMRI/JMRI #9185: JMRI forgets MERG sensors if hardware not switched on

jhitchens, 2020-11-28:
Hi guys,
I was setting up a system with PanelPro connected to CBUS MERG occupancy block detectors, and set up the sensors in the first session and got them working ok.  Later when I ran PanelPro without the hardware being present, it gives errors like 'Could not create sensor System name "MS+n16e1;-n16e1"', presumably because it can't detect the hardware, which was ok, because I wanted to make some other changes to add layout/control panel/occupancy blocks/paths etc.  Then after saving the changes and reconnecting later to MERG, I found that it had forgotten all the sensors.  Is it possible to work in PanelPro without having the MERG hardware attached without losing the MERG related sensors?
Thanks,
Jack

icklesteve, 2020-11-29:
Hi Jack,

You need 2 profiles.
1 for when hardware is attached ( MERG CAN USB / Network Interface), and 1 for without ( MERG CAN Simulation ).
See https://www.jmri.org/help/en/html/setup/profiles.shtml

The simulation can be tweaked with the CBUS Simaultion Tools, https://www.jmri.org/help/en/package/jmri/jmrix/can/cbus/swing/simulator/SimulatorPane.shtml

Steve.

jhitchens, 2020-11-30:
Hi Steve,
Thanks for the prompt feedback.  This should sort us out.
Regards,
Jack

closed 2020-11-30 by jhitchens