JMRI/JMRI #9193: Verifying multiple CMRI connections at one time

jerryg2003, 2020-12-02:
The documentation for CMRI (https://www.jmri.org/help/en/html/hardware/cmri/CMRI.shtml) seems to be written as if only a single CMRI connection can be established at one time, using the prefix "C".  Other documentation has picked up on this, such as the documentation on arduinos (https://www.jmri.org/help/en/html/hardware/arduino/index.shtml#multi).  

Several people who should know have said in two different group.io forums that this is incorrect and you can have multiple CMRI connections, "C", "C1", etc.    If this is indeed the case, has anyone verified that all CMRI functions such as the network manager, the monitor, etc. all work with multiple connections?  If so, then I'll take it as a task to update the documentation (that is, I'll add it to the list: https://github.com/JMRI/JMRI/issues/8139).  

And, by the way, does this mean sensor #4 for a second CMRI connection ("C1") using node 12 would have system name "C112004" or is this sensor #2004 on the first CMRI connection ("C") using node 11?  Or would you have to use the "alternate" numbering scheme for second CMRI connections of "C112B004" to avoid any confusion?

Thanks, Jerry

bobjacobsen, 2020-12-02:
Sensors have an “S” after the system connection prefix: CS4, C2S4, etc. There’s no ambiguity.

jerryg2003, 2020-12-02:
Bob - Of course...thanks for reminding me.  

Anyone else - know about all other CMRI tools, monitors, etc.?  Are they OK with multiple CMRI connections?

Thanks, Jerry

KenC57, 2020-12-02:
Jerry,

We think the code got a common update so it works for all systems. Somebody
who has multiple CMRI devices that can use different connections should
test. The problem is none of the developers had that, that I know of.

devel-bobm, 2020-12-02:
@KenC57 , presumably this type of configuraiton could be tested with a couple of Aduino boards, each loaded with an appropriate C/MRI sketch and connected via individual USB cables.  I believe that a user problem with such a configuration was the original motivation for this particular `Issue`.

If I had an appropriate sketch, I could test it, but I do not...

closed 2020-12-19 by jerryg2003