JMRI/JMRI #9197: JMRI does not work on a M1 Mac

dsand47, 2020-12-03:
After applying a temporary fix to StartJMRI as documented in #9151, PanelPro will not start.  It gets a Permission Denied message.  The macOS Console:
```
LAUNCH: RBSLaunchRequest FAILURE <private> jmri.PanelPro
<private> 0x0-0x2ce2ce failed with error Error Domain=RBSRequestErrorDomain Code=5
"Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x600002de2c10
{Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111"
UserInfo={NSLocalizedDescription=Launchd job spawn failed with error: 111}}}

LAUNCH: Runningboard launch of jmri.PanelPro <private> returned RBSRequestErrorFailed,
error Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed.,
NSUnderlyingError=0x600002de2c10 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111"
UserInfo={NSLocalizedDescription=Launchd job spawn failed with error: 111}}}, so returning -10826

LAUNCH: request execute thru runningboard of 0x0-0x2ce2ce jmri.PanelPro/<private> failed with error=Error
Domain=NSOSStatusErrorDomain Code=-10826 "kLSNoLaunchPermissionErr: User doesn't have permission to launch the app (managed networks)"
UserInfo={_LSFunction=_LSLaunchWithRunningboard, _LSLine=2508, NSUnderlyingError=0x600002de3ae0
{Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed.,
NSUnderlyingError=0x600002de2c10 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111"
UserInfo={NSLocalizedDescription=Launchd job spawn failed with error: 111}}}}}

handle LS launch error: {\n    Action = oapp;\n    AppPath = "/Applications/JMRI/PanelPro.app";\n    ErrorCode = "-10826";\n}
```

It is possible to start PanelPro from the command line:
```
bash-3.2$ ./StartJMRI 
PWD: '/Applications/JMRI'
./StartJMRI: line 437: readelf: command not found
./StartJMRI: line 437: readelf: command not found
./StartJMRI: line 437: readelf: command not found
CLASSPATH: '.:classes:target/classes:jmri.jar:lib/beansbinding-1.2.1.jar:lib/bluecove-2.1.1-SNAPSHOT.jar:lib/bluecove-bluez-2.1.1-SNAPSHOT.jar:lib/bluecove-gpl-2.1.1-SNAPSHOT.jar:lib/commons-compress-1.18.jar:lib/commons-csv-1.7.jar:lib/commons-io-2.6.jar:lib/commons-lang3-3.7.jar:lib/commons-text-1.2.jar:lib/gluegen-rt.jar:lib/hid4java-0.5.0.jar:lib/jackson-annotations-2.10.0.jar:lib/jackson-core-2.10.0.jar:lib/jackson-databind-2.10.0.jar:lib/javax.servlet-api-3.1.0.jar:lib/jdom2-2.0.6.jar:lib/jetty-http-9.4.28.v20200408.jar:lib/jetty-io-9.4.28.v20200408.jar:lib/jetty-security-9.4.28.v20200408.jar:lib/jetty-server-9.4.28.v20200408.jar:lib/jetty-servlet-9.4.28.v20200408.jar:lib/jetty-util-9.4.28.v20200408.jar:lib/jhall.jar:lib/jhidrawplugin.jar:lib/jinput.jar:lib/jinputvalidator-0.8.0.jar:lib/jlfgr-1_0.jar:lib/jmdns-3.5.5.jar:lib/jna-4.4.0.jar:lib/jna-platform-4.4.0.jar:lib/joal.jar:lib/json-schema-validator-1.0.28.jar:lib/jsplitbutton-1.3.1.jar:lib/jul-to-slf4j-1.7.25.jar:lib/jython-standalone-2.7.2.jar:lib/libusb4java-1.3.0-darwin-x86-64.jar:lib/libusb4java-1.3.0-linux-aarch64.jar:lib/libusb4java-1.3.0-linux-arm.jar:lib/libusb4java-1.3.0-linux-x86-64.jar:lib/libusb4java-1.3.0-linux-x86.jar:lib/libusb4java-1.3.0-win32-x86-64.jar:lib/libusb4java-1.3.0-win32-x86.jar:lib/log4j-1.2.17.jar:lib/mailapi.jar:lib/mqtt-client-0.4.0.jar:lib/openlcb.jar:lib/org-openide-util-RELEASE113.jar:lib/org-openide-util-lookup-RELEASE113.jar:lib/pi4j-core-1.2.jar:lib/purejavacomm-1.0.1.jar:lib/slf4j-api-1.7.25.jar:lib/slf4j-log4j12-1.7.25.jar:lib/thumbnailator-0.4.8.jar:lib/usb-api-1.0.2.jar:lib/usb4java-1.3.0.jar:lib/usb4java-javax-1.3.0.jar:lib/vecmath-1.5.2.jar:lib/websocket-api-9.4.28.v20200408.jar:lib/websocket-client-9.4.28.v20200408.jar:lib/websocket-common-9.4.28.v20200408.jar:lib/websocket-server-9.4.28.v20200408.jar:lib/websocket-servlet-9.4.28.v20200408.jar:lib/xAPlib.jar:lib/xbee-java-library-1.2.1.jar:lib/xercesImpl.jar:lib/xml-apis-1.4.01.jar'
Java CMD: '/Library/Java/JavaVirtualMachines/zulu-11.jdk/Contents/Home/bin/java'
APPNAME: 'PanelPro'
OPTIONS: ' -noverify -Djava.security.policy=lib/security.policy -Djava.rmi.server.codebase=file:target/classes/ -Djava.library.path=.:lib/macosx:lib -Dapple.laf.useScreenMenuBar=true -Dcom.apple.macos.useScreenMenuBar=true -Dfile.encoding=UTF-8 -Xms96m -Xmx4096m'
DOCK_OPTIONS: '-Xdock:name=PanelPro -Xdock:icon=/Applications/JMRI/PanelPro.app/Contents/Resources/PanelPro.icns'
2020-11-30 14:51:00,823 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-11-30 14:51:01.247 java[10875:704789] CoreText note: Client requested name ".SFNS-Regular", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-30 14:51:01.247 java[10875:704789] CoreText note: Set a breakpoint on CTFontLogSystemFontNameRequest to debug.
2020-11-30 14:51:01.249 java[10875:704789] CoreText note: Client requested name ".SFNS-Bold", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-30 14:51:01.281 java[10875:704789] CoreText note: Client requested name ".SFNS-Regular", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-30 14:51:01.282 java[10875:704789] CoreText note: Client requested name ".SFNS-Bold", it will get Times-Roman rather than the intended font. All system UI font access should be through proper APIs such as CTFontCreateUIFontForLanguage() or +[NSFont systemFontOfSize:].
2020-11-30 14:51:01,297 util.Log4JUtil                        INFO  - This log is appended to file: /Users/railroadm1mini/Library/Preferences/JMRI/log/messages.log [main]
2020-11-30 14:51:01,298 util.Log4JUtil                        INFO  - This log is stored in file: /Users/railroadm1mini/Library/Preferences/JMRI/log/session.log [main]
2020-11-30 14:51:01,302 apps.Apps                             INFO  - PanelPro version 4.20+Rc7ba8249b starts under Java 11.0.9.1 on Mac OS X aarch64 v11.0.1 at Mon Nov 30 14:51:01 CST 2020 [main]
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by jmri.util.swing.SliderSnap$Init (file:/Applications/JMRI/jmri.jar) to method javax.swing.plaf.basic.BasicSliderUI.xPositionForValue(int)
WARNING: Please consider reporting this to the maintainers of jmri.util.swing.SliderSnap$Init
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
2020-11-30 14:51:01,380 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3f1f4bce [main]
2020-11-30 14:51:01,421 node.NodeIdentity                     INFO  - Using 7129d3da-4b36-4146-8e89-8d619ecf1594 as the JMRI storage identity for profile id 3f1f4bce [AWT-EventQueue-0]
2020-11-30 14:51:01,548 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-11-30 14:51:01,549 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2020-11-30 14:51:03,810 util.FileUtilSupport                  INFO  - File path program: is /Applications/JMRI/ [main]
2020-11-30 14:51:03,812 util.FileUtilSupport                  INFO  - File path preference: is /Users/railroadm1mini/Library/Preferences/JMRI/My_JMRI_Railroad.jmri/ [main]
2020-11-30 14:51:03,812 util.FileUtilSupport                  INFO  - File path profile: is /Users/railroadm1mini/Library/Preferences/JMRI/My_JMRI_Railroad.jmri/ [main]
2020-11-30 14:51:03,813 util.FileUtilSupport                  INFO  - File path settings: is /Users/railroadm1mini/Library/Preferences/JMRI/ [main]
2020-11-30 14:51:03,813 util.FileUtilSupport                  INFO  - File path home: is /Users/railroadm1mini/ [main]
2020-11-30 14:51:03,814 util.FileUtilSupport                  INFO  - File path scripts: is /Applications/JMRI/jython/ [main]
2020-11-30 14:51:03,888 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 11.0.9.1 [main]
2020-11-30 14:51:03,890 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.2 [main]
Warning: Nashorn engine is planned to be removed from a future JDK release
[... cut ...]
2020-11-30 14:51:39,514 jmrix.AbstractSerialConnectionConfig  ERROR - UnsatisfiedLinkError - the serial library has not been installed properly [AWT-EventQueue-0]
2020-11-30 14:51:39,514 jmrix.AbstractSerialConnectionConfig  ERROR - java.library.path=.:lib/macosx:lib [AWT-EventQueue-0]
```

The `readelf` messages are references to a Linux command which does not exist in macOS.

The `lib/macosx` error is caused by the Intel code in the jnilib and dylib files.  Apparently Rosetta does not handle Intel code called within the Java context, which is running an Arm version of openjdk.

dsand47, 2021-01-15:
Feedback from Azul in response to query from Jim LeFevre.

Eduardo Ahumada (Azul Systems, Inc.) Jan 14, 2021, 20:16 EST

Hello Jim,

Our engineering team points to contact the JRMI team to fix the problem, as it the root cause is not in the Zulu JVM, but in the JRMI libraries used that are not compiled for M1.

It seems purejavacomm is the serial port manager for this project, the library should be rebuild for ARM64 arch. 

We I've checked JMRI 4.20. It contains native libraries apparently used for accessing hardware, like ./JMRI/lib/macosx/libjinput-osx.jnilib. It does not support arm64 architecture, so it cannot be used with arm64 java.

Using x86-64 java on m1 mac with rosetta translator could be a workaround.

But actually, the author of JMRI should compile the code to add support for m1 Mac.

---

The response included the **_file_** output but I used my own since it was easier to include.

```
dmba:macosx das$ file libjinput-osx.jnilib 
libjinput-osx.jnilib: Mach-O universal binary with 3 architectures: [ppc:Mach-O dynamically linked shared library ppc] [i386:Mach-O dynamically linked shared library i386] [x86_64:Mach-O 64-bit dynamically linked shared library x86_64]
libjinput-osx.jnilib (for architecture ppc):	Mach-O dynamically linked shared library ppc
libjinput-osx.jnilib (for architecture i386):	Mach-O dynamically linked shared library i386
libjinput-osx.jnilib (for architecture x86_64):	Mach-O 64-bit dynamically linked shared library x86_64
```

---

If Rosetta is enabled and a x86-64 Java is installed, then the existing libraries work.

mattharris, 2021-01-15:
I think that we need to wait for https://github.com/java-native-access/jna/pull/1297 to be resolved as JNA is used for the underlying hardware access by purejavacomm.

dsand47, 2021-01-21:
> @dsand47 with regard the `readelf` message, what is the output from the following shell commands:
> 
> ```
> arch
> uname -i
> uname -p
> uname -m
> ```
> 
> Results of those are used to determine the underlying architecture of the machine and, right now, we have some special handling of certain ARM systems to handle the different hard-float/soft-float and instruction set combinations seen with ARM-based Linux machines which, until now, would not have been acted upon with macOS.

I don't have a M1 Mac.  We will need to get that information from Jim LeFevre.

He is currently using a x86-64 JDK which invokes Rosetta which then handles the lib components.  Using the Arm version of the JDK did not work because the lib components where being handled by Java so Rosetta was not invoked.

dsand47, 2021-01-22:
M1 terminal command results:
```
railroadm1mini@Railroad-M1-Mini ~ % arch
arm64
railroadm1mini@Railroad-M1-Mini ~ % uname -i
uname: illegal option -- i
usage: uname [-amnprsv]
railroadm1mini@Railroad-M1-Mini ~ % uname -p
arm
railroadm1mini@Railroad-M1-Mini ~ % uname -m
arm64
```

jszinger, 2021-02-03:
The following files are missing arm64 on Mac OS support:

```
$ file lib/macosx/*
lib/macosx/libgluegen-rt.jnilib:   Mach-O universal binary with 2 architectures: [i386:Mach-O i386 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>] [x86_64:Mach-O 64-bit x86_64 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>]
lib/macosx/libjinput-osx.jnilib:   Mach-O universal binary with 3 architectures: [ppc:Mach-O ppc dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>] [i386:Mach-O i386 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>] [x86_64:Mach-O 64-bit x86_64 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>]
lib/macosx/libjoal.jnilib:         Mach-O universal binary with 2 architectures: [i386:Mach-O i386 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>] [x86_64:Mach-O 64-bit x86_64 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>]
lib/macosx/libopenal.1.15.1.dylib: Mach-O 64-bit x86_64 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>
lib/macosx/libopenal.1.dylib:      Mach-O 64-bit x86_64 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>
lib/macosx/libopenal.dylib:        Mach-O 64-bit x86_64 dynamically linked shared library, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|NO_REEXPORTED_DYLIBS>
```

dsand47, 2021-03-30:
This issue is dependent on release of M1 compatible third party components.

dsand47, 2021-05-14:
This issue is still dependent on release of M1 compatible third party components.

bobjacobsen, 2021-05-14:
I'm working on it, will post occasionally about progress.

Right now, there's an installer at https://builds.jmri.org/jenkins/job/development/job/m1_test_packages/ that uses updated libraries, but doesn't work.  The bugs seen have been fed back upstream.

bobjacobsen, 2021-06-28:
Still working on this.

bobjacobsen, 2021-08-12:
I'm still working on this with the PJC group.

bobjacobsen, 2021-09-27:
Still working on this.  PJC is having trouble with certain configurations.

bobjacobsen, 2021-11-11:
Status of current development builds:  
 - Basic install and operate OK in M1-only mode
 - JOAL used by audio does not yet have native support, works with Rosetta 2
 - LibUSB used by Anya lighting support does not yet have native support, works with Rosetta 2
Work continues.

geowar1, 2021-11-12:
According to [https://doesitarm.com/formula/libusb/](url) libUSB has native support.

bobjacobsen, 2021-11-13:
We distribute the components for using libUSB with JMRI itself:

% ls lib/libusb*
lib/libusb4java-1.3.0-darwin-aarch64.jar
lib/libusb4java-1.3.0-darwin-x86-64.jar
lib/libusb4java-1.3.0-linux-aarch64.jar
lib/libusb4java-1.3.0-linux-arm.jar
lib/libusb4java-1.3.0-linux-x86-64.jar
lib/libusb4java-1.3.0-linux-x86.jar
lib/libusb4java-1.3.0-win32-x86-64.jar
lib/libusb4java-1.3.0-win32-x86.jar

There is an aarch64 jar there, and it contains the right architecture:

% jar tf lib/libusb4java-1.3.0-darwin-aarch64.jar
META-INF/
META-INF/MANIFEST.MF
org/
org/usb4java/
org/usb4java/darwin-aarch64/
org/usb4java/darwin-aarch64/libusb4java.dylib
org/usb4java/darwin-x86-64/
org/usb4java/darwin-x86-64/libusb4java.dylib

It seems to work partially, in that you can go to PanelPro Debug -> USB Devices and get a USB Browser.  But somehow the Anyma support isn’t working, with a crash in the USB code during startup.

Maybe we’ve got the wrong version of libUSB? The Homebrew page as 1.0.24 is current, but the file names seem to indicate we’re using 1.3.0.  The lib/README says:

```
##### usb-api-1.0.2.jar, usb4java-*.jar, libusb4java-*.jar
- usb4java version 1.3.0
- support for direct USB device usage
- from https://github.com/usb4java/usb4java/releases/tag/usb4java-1.3.0
  and https://github.com/usb4java/usb4java-javax/releases/tag/usb4java-javax-1.3.0
- lib/libusb4java-1.3.0-darwin-aarch64.jar made from a .dylib found at https://github.com/developersu/ns-usbloader/issues/91
```

Which is a bit sketchy.

geowar1, 2021-11-13:
The error JMRI logs when preferences attempts to select a Anyma connection says that it can't find/load libusb4java.dylan at /org/usb4java/darwin-aarch64 (which is a directory/file that doesn't exist).

Sent from GeoWar’s iPad Pro

bobjacobsen, 2021-12-28:
Still active, waiting to get some more library updates for libusb4java and JOAL.

klk32003, 2021-12-29:
There is a first contribution on JOAL, see https://forum.jogamp.org/JOAL-for-MAC-arm64-td4041564.html

If I see this correctly the following three native files are to be exchanged in jmri/lib/macosx:
- ```libgluegen_rt.dylib``` - see https://github.com/maheshkurmi/jogl-macarm64-builds/blob/main/gluegen-rt-natives-macosx-universal.jar
- ```libjoal.dylib``` and ```libopenal.dylib``` - see https://github.com/maheshkurmi/jogl-macarm64-builds/blob/main/joal-android-natives-macosx-universal.jar

JUnit tests are not complete, but a recipe is documented how to do the build of the universal native files. So a test might be useful.

bobjacobsen, 2022-02-12:
still waiting on updates to three libraries.

closed 2022-07-24 by stale[bot]