JMRI/JMRI #9210: CMRI create sensors >999 on one node fails

KenC57, 2020-12-08:
When trying to create more than 999 inputs on a node it fails with:
[image]
Creating turnouts in that range work ok.

icklesteve, 2021-04-01:
Looks like this is hard coded into CMRI SerialSensorManager, 

CMRI SerialSensorManager defines 
  `static final int SENSORSPERUA = 1000;`
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/cmri/serial/SerialSensorManager.java#L40

CMRI SerialNode may also limit the number of Sensors to 999, see
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/cmri/serial/SerialNode.java#L52

bobjacobsen, 2021-04-01:
There's a bunch of code to handle the CT1B1001 format for system names, but it causes problems in various downstream things that have never been resolved.

I expect there'd be more enthusiasm for all that work if somebody actually had hardware with 1000 input or output pins.  I don't think the real C/MRI hardware can do it, but perhaps somebody has a pseudo-C/MRI Arduino implementation that can.

closed 2021-06-01 by stale[bot]