JMRI/JMRI #9211: CMRI List Assignments tool - incorrect address when >999 turnouts or sensors on a node

KenC57, 2020-12-08:
The bit number and system name both look fine. But the address field seems to have messed up the node part above 999 bits.
[image]
This effects both the input and output bits. But I'm not sure what the address column is trying to show us. There is no meaningful help page for this tool.

silverailscolo, 2020-12-08:
This seems to describe the goal and use of the tool: [CMRI help](https://www.jmri.org/help/en/html/hardware/cmri/CMRI.shtml#L609) (in the nav block, click list I/O Bit Assignment).

KenC57, 2020-12-08:
I think some of the text on the CMRI Help page would be useful on the window help file for this tool. It looks like a lot of the work that Chuck did put all the documentation in the one help file and not spread around to the many tools. I would think having them in the file for each tool would be much easy to find than in the one rambling file. Or leave it there and move parts into the many tools.

jerryg2003, 2020-12-10:
See my updated version of the CMRI help page for further discussion on >999 numbering.  PR #9209 

Jerry

closed 2021-04-14 by stale[bot]