JMRI/JMRI #9213: Renamed file under both names on the web site

jerryg2003, 2020-12-10:
Some months ago, I renamed a file in the Git repository from https://www.jmri.org/help/en/html/apps/SaveWhatHowWhereWhen.shtml to https://www.jmri.org/help/en/html/apps/DataManagement.shtml .  See PR #8637. 

Only the latter is now in the Git repository, as it should be.  However, both seem to exist and are accessible on the web (the former comes up in a google search).  Why?

Other files like this where name changed and old file didn’t disappear?

Thanks, Jerry

danielb987, 2020-12-10:
I don't know. But I suppose files are copied to the web server and therefore not removed when deleted from git.

@bobjacobsen or @mattharris might explain this.

jerryg2003, 2020-12-11:
Somewhat of a concern since old file is indexed and comes up in various searches, usually ahead of the new version.  Confusing.

bobjacobsen, 2020-12-11:
"Somewhat of a concern since old file is indexed and comes up in various searches". That's exactly why the old .shtml file is kept.  Otherwise, a large number of dead links from search engines and bookmarks can be created.

If all the internal links are changed to the new file, eventually the search engines will drop the old file.  But that can take a long time if there are still links to the old file.

If this is a problem, one solution is to put a forwarding file in place that points to the new location.

jerryg2003, 2021-01-11:
So I have for this particular case, recreated the file in Git to change its contents to just be forwarding text (this will be in my next PR). Once out on the web, I will delete the old-named version in Git so that no one tries to work with it.

In other cases I have now found, there are files on the web server that are just taking up space.  For example, I see that at least some (presumably all) of the old "Sidebar" files are out there even though we have replaced them with "Sidebar.shtml" files.  There are no references to these old files and they don't exist in the Git repository, so they could clearly be deleted without harm.  Also found some other case of renamed files (which I did for clarity - or so I thought) being out there in addition to the newer named versions even though they are useless (and not in Git).  Don't know if any other people have done any renaming but likely both versions of these exist on the server as well.

Would it be useful to delete and rebuild the web server contents (at least for "/help"), perhaps once a year or when a new higher-order numbered release comes out (say 5.0?) so that what is physically on the server corresponds to what is in Git?  Seems like that is where we should be to avoid any issues in the future.

bobjacobsen, 2021-01-11:
We can certainly periodically clean the web site of non-Git files.

If we’re going to do that, we should probably start doing it with every update, though.  

I don’t know what kind of issues these files might cause for users, but if there are any then we should probably do the clean right away (every update) instead of later.

Bob

> On Jan 10, 2021, at 10:30 PM, JerryG <notifications@github.com> wrote:
> 
> Would it be useful to delete and rebuild the web server contents (at least for "/help"), perhaps once a year or when a new higher-order numbered release comes out (say 5.0?) so that what is physically on the server corresponds to what is in Git? Seems like that is where we should be to avoid any issues in the future.

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

danielb987, 2021-01-11:
One issue users might have is bookmarks. If I find an interesting page, I sometimes bookmark that particular page.

So if we have a page https://www.jmri.org/help/en/html/doc/Technical/Threads.shtml that is moved to https://www.jmri.org/help/en/html/doc/Technical/AllAboutThreads.shtml, it's good practice to keep the old file on the server and have a forward to the new file.

But for internal files, like css files and sidebars, this should not be a problem.

jerryg2003, 2021-01-11:
Again, I see the point about old links. OTOH, it makes it difficult to do any reorganization and  cleanup, say when a directory becomes unwieldy, as for example /jython, and there could be benefit from having some files organized by topic into sub-directories.

maybe three cases here:
1. “Helper” files like css, sidebar, header/footer,  parts, etc. where old files no longer in Git should be deleted from server.
2. Help content files where instead of renaming, files should be duplicated with new name and old name file should have its contents replaced with redirect code. [this requires on-going cooperation from people working on the documentation.]
3. /jython, /web, /xml and perhaps others where files not in Git should be deleted from server.

Eh?

danielb987, 2021-01-11:
1. All helper files can be deleted without problems.
2. Duplicate is better than renaming. The old file doesn't need to keep the content, just a link to the new page.
3. Not sure about these.

silverailscolo, 2021-01-12:
Just to specify a "link to the new file": instead of a replacement with a (user clickable) hyperlink, these are non-displaying so called redirect files, which immediately get the browser display the new file.
In cases like the ``jython`` folder, at least make sure a file named ``index.shtml`` is present, serving as the default for the directory url.

jerryg2003, 2021-01-19:
If we put an index.shtml into /jython, then no one will be able to get a list of the files in the directory, which you get now by just going to jmri.org/jython in your browser.  This also displays the contents of HEADER.html which currently has a couple lines of text and a link to JMRI scripting help.  I think the current approach may be more useful than having an index.sthml since individual files are often referenced in the user forum and we can always add some more text to the header.

bobjacobsen, 2021-01-20:
You could do something like the index.php in

https://www.jmri.org/resources/

It contracts a useful display.

Bob

> If we put an index.shtml into /jython, then no one will be able to get a list of the files in the directory, which you get now by just going to jmri.org/jython in your browser. This also displays [...]

—
Bob Jacobsen
rgj1927@gmail.com

jerryg2003, 2021-03-13:
For now, files will be deleted by individual request to someone with access to the web server.

closed 2021-03-13 by jerryg2003

icklesteve, 2021-03-13:
Is this also the cause of https://github.com/JMRI/JMRI/issues/8676 ?

jerryg2003, 2021-03-13:
I’m not familiar with issues relating to the java code.  However, if the server update process works the same way, then copies of files that were moved or renamed continue to exist unless/until someone deletes them manually.