JMRI/JMRI #9215: CPE - Warrants - web server - incorrect display of turnout indicator status

sidlo64, 2020-12-10:
Used icons style - icons/panels/CSD/JOP
The panel window shows the status for the Turnout Indicator for the Running Warrant in Magenta color.
The Turnout Indicator on the phone screen is in Red color - Occupied Track. It should be Magenta - Warrant.
Missing function - the train name is not displayed in the web server.

[image]

[image]

mstevetodd, 2020-12-10:
@sidlo64 please attach recreation materials, like a save of your config (set to autoload the panel, etc.) and any manual steps needed to reach the problem described.

sidlo64, 2020-12-10:
How to Reproduce
unzip WebCPE.jmri
Load panel CPE
Click 6 buttons in 1st row, make all block unoccupied
Click 1st button, make 1st block occupied
Set turnouts to Closed
Start JMRI Web Server
Open browser
Open Warrant Table
Auto Run Warrant <code>#1</code>
Resume Control/Status
Click 2nd and 3rd buttons
See result
Turnout should be Blue but is Yellow

[WebCPE.jmri.zip](https://github.com/JMRI/JMRI/files/5672795/WebCPE.jmri.zip)

[image]

silverailscolo, 2020-12-10:
I created the web CPE and can look at this issue. First item you describe concerns (only) the turnout icon, correct? Those icons are more complex than simple tracks, as they have both a state (position) and a status (Warrant). @sidlo64 Can you tell the int value of the Status in this case?
I decided to have Train Name as an option not yet supported. Same goes for combo display.

sidlo64, 2020-12-10:
I encountered this problem when I was doing the widget for the level railroad crossing. It is at first JOP panel in the middle. The problem was that OBlock can have multiple states concurrently. When the track/turnout indicator is displayed, some states take precedence over others. I couldn't use the turnout widget and for railroad level crossing wrote my own script. Try script turnout.py.

[turnout.py.zip](https://github.com/JMRI/JMRI/files/5674571/turnout.py.zip)

Open Script Output and then Run Script...

closed 2020-12-15 by silverailscolo