JMRI/JMRI #9217: Possible error in AAR-196 aspect logic

bobjacobsen, 2020-12-10:
(From a jmriusers thread where a user noticed multiple yellow appearances in a row)

There might be a bug in the AAR-1946/appearance-SL-1-high-pbs.xml and AAR-1946/appearance-SL-1-high-abs.xml  files, which have:
```
     <aspectMapping>
     <advancedAspect>Approach</advancedAspect>
     <ourAspect>Approach</ourAspect>
     </aspectMapping>
```
In their mappings. (It’s actually there _twice_)  Does anybody know if there’s a reason for this to be there? They’ve been in the file for ages.

dsand47, 2020-12-10:
Bob,

It might be there to support the following:
Therefore the circuitry is designed to always provide two signals displaying "Approach" where a train might be approaching from the other side of the siding. This is often called *double yellow*, referring to the yellow light of a color light signal displaying "Approach".

http://www.lundsten.dk/us_signaling/abs_st_sp/p_index.html

Dave Sand

devel-bobm, 2020-12-10:
It looks kinda like what one might try to implement “double-approach” signaling.  But it won’t be a very good solution…

bobjacobsen, 2020-12-10:
I don't think double-approach is in the 1946 rule set.  At least, I don't remember seeing it there.  It was mostly an SP thing.

I think this is just a data-entry mistake. It hasn't been seen before because most model railroads aren't big enough to have multiple block signals between sidings.

devel-bobm, 2020-12-15:
"Double-approach" is not "part of a rule-set".  Instead it is a term-of-art which describes the provision of two signals in sequence to both present the "Approach" indication to protect a third signal at "Stop".  

"Double-approach" can be a key part of protecting the approaches to a siding in single-track signaled for both directions in APB territory.  The attached diagram and text is from a 1935 AAR circuits document, and describes the need for two Approach indications in sequence, in the last sentence of the paragraph.
[image]
[image]

Diagram from page 33 of [http://rrsignalpix.com/pdf/AAR_15_sec.pdf](url), text from page 34.

closed 2021-04-14 by stale[bot]