JMRI/JMRI #9219: Fresh install - doesn't run :-(

Steve058, 2020-12-11:
Hi, I have just done a fresh installed of JMRI (4.20) on a Win10 machine. Machine already had JAVA so the install didn't download that.
But when I try to run either DecoderPro or PanelPro nothing happens.
There is nothing in Eventvwr. If I run from CMD prompt, no errors are reported.

Any ideas what I can look at next?

Here's my CMD prompt result -
`Microsoft Windows [Version 10.0.19041.630]
(c) 2020 Microsoft Corporation. All rights reserved.

C:\Users\Steve>java -version
java version "11.0.8" 2020-07-14 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.8+10-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.8+10-LTS, mixed mode)

C:\Users\Steve>"C:\Program Files (x86)\JMRI\LaunchJMRI.exe" apps.gui3.dp3.DecoderPro3

C:\Users\Steve>`

klk32003, 2020-12-12:
Next you can do is to start "InstallTest" from the JMRI group in your Start menu, see https://www.jmri.org/install/WindowsNew.shtml#debugging

Please post the result here.

Steve058, 2020-12-12:
OK, I've run the InstallTest. The CMD window part says -
`Testing for 32/64-bit Windows
Running on 64-bit Windows
Testing for 32/64-bit process
Running as a 64-bit process
Testing for Java working
java version "11.0.8" 2020-07-14 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.8+10-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.8+10-LTS, mixed mode)
Java is correctly working.
Press any key to continue . . .
Now testing JMRI
Something is wrong with invoking JMRI. Check JMRI installation.
Press any key to continue . . .
Test complete
Press any key to continue . . .`

and the popup window said -
[InstallTest.txt](https://github.com/JMRI/JMRI/files/5683029/InstallTest.txt)

klk32003, 2020-12-12:
Thanks.

The Java version in the CMD window (11.0.8) and the Java version in the InstallTest.txt (1.8.0_241) differs.

You could try setting JAVA_HOME or deinstall one Java version.

klk32003, 2020-12-12:
Another thought. JMRI cannot find Java 11 because the Windows registry key doesn't match. See the [code](https://github.com/JMRI/JMRI/blob/master/scripts/WinInstallFiles/LaunchJMRI.nsi#L302) for the search strategy.

Steve058, 2020-12-12:
Hi, well spotted. I didn't see that.
When I went to Control Panel / Java to set the default, it showed me a blank icon and said it couldn't find the application. So something wrong with my Java setup.
I ended up uninstalling version 1.8.0 and hey-presto JMRI starts working :-)
So I just need to remember what other app needed Java and see if that still works, but that's not your problem.

Thanks for the help.

closed 2020-12-12 by Steve058