JMRI/JMRI #9220: PanelPro Tables view, bug

MarcFour, 2020-12-12:
Resizing the PanelPro ,tables view  I will loose the left hand inner margin. Unable to recover no matter what I attempt

What I see is in user-interface.xml  that the LocX and LocY  values end up on -8.0

<window xmlns="" class="jmri.jmrit.beantable.ListedTableAction$1$1:Sensors" height="1000.0" locX="-8.0" locY="-8.0" width="1296.0"/>

        <window xmlns="" class="jmri.jmrit.beantable.ListedTableAction$1$1:Turnouts" height="1000.0" locX="-8.0" locY="-8.0" width="1296.0"/>

Try to correct this by deleting the line above but I end up again with : 

<window xmlns="" class="jmri.jmrit.beantable.ListedTableAction$1$1:Turnouts" height="564.0" locX="0.0" locY="31.0" width="1280.0"/>

[image]

Marc N Fournier
Quebec

MarcFour, 2020-12-12:
Found a post where the poster described encountering same issue as I am seeing.
https://groups.io/g/jmriusers/message/156233

Only solution I found to recover the left hand field was by deleting all of user-interface.xml.

sidlo64, 2021-02-08:
Less dirty trick 
file `user-interface.xml`
edit integer value
```
<properties>
    <property>
        <key>dividerLocation</key>
        <value class="java.lang.Integer">26</value>
    </property>
</properties>
```

silverailscolo, 2021-02-25:
Fixed in 4.21.5, closing

closed 2021-02-25 by silverailscolo