JMRI/JMRI #9234: Warrant - saved throttle commands in bad format

sidlo64, 2020-12-13:
Warrant created in Prototypical mode and saved with panel.
After next PanelPro start it is not readable due to error

`2020-12-13 18:53:33,555 configurexml.ErrorHandler             ERROR - Parse error while parsing file /home/tata/.jmri/CPE9NASIM.jmri/panel.xml Exception: org.jdom2.input.JDOMParseException: Error on line 5357: cvc-datatype-valid.1.2.1: '9.611761E-4' is not a valid value for 'decimal'.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 5357: cvc-datatype-valid.1.2.1: '9.611761E-4' is not a valid value for 'decimal'.
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:232)
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:253)
...
`

code in panel:
`      <throttleSetting elapsedTime="4756" beanName="OB:AUTO:0004" trackSpeed="9.611761E-4">
        <command commandType="1" fKey="-1" />
        <commandValue valueType="1" speedMode="unknown" floatValue="0.007936508" />
      </throttleSetting>
`

[CPE9NASIM.jmri.zip](https://github.com/JMRI/JMRI/files/5685127/CPE9NASIM.jmri.zip)

[session.log](https://github.com/JMRI/JMRI/files/5685126/session.log)

silverailscolo, 2021-01-31:
From W3Schools:

Scientific Numbers
A floating point number can also be a scientific number with an "e" to indicate the power of 10:

Example
float f1 = 35e3f;
double d1 = 12E4d; <-- stored
System.out.println(f1);
System.out.println(d1);

Is JMRI .xml storage missing the d notation on the end? (very slow crawling speed BTW)

bobjacobsen, 2021-01-31:
XML's `decimal` type doesn't allow E notation.  It's just digits.digits.

bobjacobsen, 2021-01-31:
Not sure why the link didn’t come through on that last comment.

https://www.w3.org/TR/xmlschema-2/#decimal

> 
> 
> XML's decimal type doesn't allow E notation. It's just digits.digits.

—
Bob Jacobsen
rgj1927@gmail.com

silverailscolo, 2021-02-02:
Studying the link, one paragraph down, for _float_ (our case here) it says:
`For example, -1E4, 1267.43233E12, 12.78e-2, 12 , -0, 0 and INF are all legal literals for float.`
Instead of preventing E notation being written to xml (by rounding or so) I suggest to make sure such a string is properly interpreted.
Currently `warrants-2-9-6.xsd` defines this value as `xs-decimal`. Same applies to `commandValue - floatValue`, also handled as a float in the code.
Will open a PR shortly.

bobjacobsen, 2021-02-02:
`float` and `double` XML types are different.  The error is saying that the value is defined as an XML decimal type.  

You can fix that by making sure the code write an appropriate format for XML decimal, or by changing the schema to an XML float.

bobjacobsen, 2021-02-02:
I may be interpreting the error message wrong, but I think this is a mismatch between the in-code type and the XML type.

sidlo64, 2021-02-03:
I came to the conclusion that this is a pseudo-problem.
The number is stored in E notation if it is less than 0.001.
Throttle openings of less than 0.001 should never occur in the speed profile.
Such a number will only get into the speed profile if:
- an automatic merge profile is used
- and at the same time the speed profile is calculated incorrectly.
In my experience, the profile will be calculated incorrectly in the case of jitter on the occupancy sensor when debounce is not used.
In my opinion, the cause of the error is somewhere in the calculation of the profile from the values of normal operation.

silverailscolo, 2021-02-04:
I saw that your case showed on a very small value, see initial comment. But the number format inconsistency has now been fixed by making the XML schema match the value type in the code, see PR reference above.
@sidlo64 if you wish to warn the user or silently round to 0 the actual value in the code to account for the sensor jitter, that seems to be a separate issue, that you could address in the Java code (of the profiling tool?)

silverailscolo, 2021-02-07:
Calculation contin. as issue #9390. Closing this now hat value type fix has been merged.

closed 2021-02-07 by silverailscolo