JMRI/JMRI #9244: LocoNet Duplex Group Setup tool does not recognize UR93

devel-bobm, 2020-12-19:
For LocoNet systems with one or more UR93, but no UR92s, the Duplex Group Setup tool cannot be used to view or change the Duplex Group's configuration.  If at least one UR92 is present, the tool will allow viewing and configuring the Duplex Group settings.

devel-bobm, 2020-12-19:
"Configure Duplex Group" is the LocoNet menu item name.

KenC57, 2022-01-19:
I've got some samples of data from a system with a UR93. I hope this might shed light on what prevents JMRI from dealing with them. This file is what happens when I do a query from the Duplex tool.

[JMRIDupGroup-A.txt](https://github.com/JMRI/JMRI/files/7893571/JMRIDupGroup-A.txt)

This next file is what happens when I use the query in the Group too.
[JMRIQuery-B.txt](https://github.com/JMRI/JMRI/files/7893581/JMRIQuery-B.txt)

Next is what happens when I plug in a UR93 and power it up.
[UR93-A.txt](https://github.com/JMRI/JMRI/files/7893584/UR93-A.txt)

KenC57, 2022-01-19:
I had two different UR93's one was 3A and the other was 3B. All other bytes in the replies were the same.

KenC57, 2022-01-19:
First step is find out what JMRI is looking for that doesn't see a UR93 the same as it sees a UR92. Seems like
a simple OR case to me.

devel-bobm, 2022-01-19:
@KenC57,

I have understood that since the day after I got my UR93.  That part is _easy_.  But adding "recognition" of the UR93 to the Duplex Group Configuration tool would open up the _huge_ can of worms that is UR91/UR92/UR93/LNWI compatibility.  I recognized that fact on that same day, and I decided that day that I _would not_ make the mistake of simply adding "recognition" of the UR93 to the tool.  

I've given this _lots_ of thought over the intervening months, and have not found a good solution that will work for all combinations of users of the variously-affected Digitrax hardware.

KenC57, 2022-01-19:
Are you are worrying about what might be native incompatibility of the Digitrax hardware? I suspect they don't
handle it very well either. At least that's how it seems trying to use the Digitrax software, it has issues
too. Plus it seems that the Digitrax software somehow has issues working with some devices like the LocoBuffer
most of the time. It leaves very awkward ways to try to get things done.

 In light of which hardware is no longer in production I find it a reason to question the value of being
concerned about those devices. Unless some of this concern is based on Digitrax documentation, we're dealing
with reverse engineered work. Going for 'it seems to work' is how a lot of JMRI features got built in the
first place. I've not heard of Digitrax releasing any new documentation about the LocoNet in years.

devel-bobm, 2022-01-19:
> Are you are worrying about what might be native incompatibility of the Digitrax hardware?

No.  

- I am concerned, in part, about _nuances_ and situations where the hardware _does not do_ what the user intends, but does something entirely different, because of hardware limitations, _without_ informing the user that it has done something different.  

- I am concerned, in part, about JMRI's ability to _know_ what the hardware has decided to do.  

- I am concerned about becoming the front-line "help desk" for Digitrax simply because Digitrax changed (for good reasons!) the features, in sometimes subtle ways.  I dread the user request "It always used to work this way, but it doesn't any more.  You need to fix JMRI!".  I further dread having to justify JMRI implementations in the environment of an extremely-complicated functional environment.

>  In light of which hardware is no longer in production I find it a reason to question the value of being
concerned about those devices.

Can we drop support for _all_ of the out-of-production hardware, across all manufacturers?  That would make JMRI "unusable" by many of our users.  Some Digitrax examples I can think of off the top of my head: Dozens of mobile decoders, DS54, DS64, PM4, BDL16, BDL162, PM4, PM42, DB150, DCS100, DCS200, DCS50, DCS51, DT100, DT200, DT300, DT400, DT402, DT500 (I think), UR90, UR91, UR92, ...  While dropping support for some of them might not ever be noticed, dropping others _would_ be noticed by existing users, including some who are interested in this specific Git issue.

> Unless some of this concern is based on Digitrax documentation, we're dealing
with reverse engineered work.

As one of the more-prolific Digitrax/LocoNet "reverse-engineers" in the JMRI developers team, I am well aware of that.  But I am _also_ well-aware of several publicly-known limitations, many of which are _documented by Digitrax_ in documentation shipped with their products.  Many users do not understand the implications of those limitations.  And I have reverse-engineered (discovered, really) other "undocumented" limitations.

Perhaps my knowledge makes me too conservative with respect to this issue.  So be it.

devel-bobm, 2022-05-02:
closed by P/R #10663 .

closed 2022-05-02 by devel-bobm