JMRI/JMRI #9252: Access SV2 Discovery and LNCV programming

silverailscolo, 2020-12-20:
In addition to a UI to discover LocoNet SV2 modules, working on support for LNCV programming of stationary devices from Uhlenbrock, Digikeijs.
(I first tried SV2 to set up a DR5033 booster, before I found out why it didn't answer...)
LNCV 15 byte messages show up interpreted in the LocoNet Monitor. Opening and closing an LNCV programming session asks for a tool in the LocoNet menu.
Coding will take another month or so, but announcing it as an Issue so it shows up in a search.
Egbert

devel-bobm, 2020-12-20:
@silverailscolo,

Be aware that I am actively working on SV2 device "discovery" and "matching discovered devices to decoder definitions" and "matching discovered devices to roster entries".  I have a GUI tool for performing discovery and reporting results in a table, with pushbuttons for "SV2 change address" and "SV2 reconfigure/reset" options.
Just today I have working (but not yet cleaned-up!) code that makes it possible to identify a roster entry that corresponds to a discovered device.

I do not have knowledge of LNCV mechanism specs.  the little bit I have in my notes suggests that LNCV is not well-defined amongst devices supporting it, such that there may be conflicts with other well-defined and well-behaved LocoNet messages.  As such, I do not intend to work on coding related to LNCV programming.

silverailscolo, 2020-12-21:
@devel-bobm,
Thanks for the heads up. I already planned to ask you to review what I [prepared](https://github.com/silverailscolo/JMRI/tree/eb-sv2discoverpane/java/src/jmri/jmrix/loconet/swing/lncvprog). My solutions for SV2 and LNCV are quite similar, as is the structure of the LNCV message support. How to link to the DecoderPro programmer interface I haven't yet figured out. Received some code for SV2 discovery from @danielb987, who had parked it too.

The 15 byte (0x0f) length of LNCV messages make them unique in JMRI LocoNet support, but you are correct to point out that this fact is not preventing anyone of coppying that for other applications. Still, I buy this box and hope to access it.

Given your effort, I will focus on LNCV as I expect users might hope to find a good tool for both (and I have an actual LNCV booster to set up). If I can assist you in checking or i18n of the SV2 tool, just let me know.

devel-bobm, 2020-12-21:
@silverailscolo,

My current code is at https://github.com/devel-bobm/JMRI/tree/dbm-fixDeveloperID .  When run, the SV2 tool is located in the LocoNet menu, with an obvious name.  

Current features:
- Push the button at the bottom to request "SV2 discovery".  The table gets updated based on info reported by SV2 devices.  
- Each device can have its address changed or receive a "reconfigure/reset" SV2 message, based on push-button activation on its row.
- After discovery, each device with non-zero destination address value gets read for SV2 value (software version, per the SV2 spec).  If multiple devices have identical destination address, then chaos ensues - I have not coded to check for this problem yet.  This value _could_ be used in decoder definition selection, but I have not yet figured out the required "hooks" into the various Roster and decoder files.  This would be very useful for any SV2 device which has had an ever-growing feature-set across different software versions (such as the CV-based, I think, LocoIo board).
- Each row's data is compared against known-SV2-capable decoder definitions.  If one is found with matching manufacturerID, developerID, and productID values in the definition's "<model>" element, its name is shown.  Only one matching name is shown.
- Each row's data is compared against "locos" in the roster.  If a roster entry has matching manufacturerID, developerID, productID, and destination address values, the roster entry's "ID" value is displayed.
- Table column widths and window position are persisted.

Near-term goals:
- be able to open a "symbolic programmer" for the row's "roster entry" from the tool;
- figure out how to reduce the default number of rows in the table, and shrink/grow the # rows visible as the window is resized;
- Move the "discover" button _outside_ of the scroll-box.
- ensure that I haven't broken programming track "identify loco" process

Stretch-goals:
- a feature which, when activated, will look for "destination address" collisions, and advise to change one or more device's address
- a feature which, when activated, will find those devices with "destination address" of 0 and sequentially request new addresses and perform the address change.

I do not see a need to "persist" any of the data in the table, as it largely can be "persisted" in roster entries.  So there won't be any sort of "save" feature.

devel-bobm, 2020-12-21:
[image]

devel-bobm, 2020-12-23:
Further progress gives the basic functionality I was trying to achieve.

[image]

Next to check that I haven't broken "identify decoder" functionality for non-SV2 decoders...

silverailscolo, 2020-12-27:
I can report progress on the LNCV tool and protocol. Got specs from Uhlenbrock and success testing on hardware.

silverailscolo, 2020-12-28:
The LNCV tool (without the table for now) is required to open an LNCV "Programming Session":
[image]
I could read values outside of a session (like temperature) but get a 306 timeout talking to CS trying to write.
For incidental programming the tool allows reading and writing of individual CV value.
With a session (to a specific device) running, you can use the Roster entry programming panes. I added a definition for the Digikeijs DR5033 Booster, via LNCV programming mode and 5033.8 style CV numbers:
[image]

silverailscolo, 2021-01-02:
@devel-bobm as I completed the LNCV interfave, I added several Bundle keys that you might also use in the LNSVf2 Discovery tool, see [LocoNetBundle starting line 30](https://github.com/silverailscolo/JMRI/blob/553cf02f36ffbead20c45a3129dfd1bef273c390/java/src/jmri/jmrix/loconet/LocoNetBundle.properties#L30), including the error feedback strings. Current UI proposal:
[image]
I chose to use column names that matches those in the Roster (see image). When you click a row in the table, the module article number and known address will be filled in at the top.