JMRI/JMRI #9264: Panel file name not available after loading (4.21.3)

jerryg2003, 2020-12-30:
[Issue copied from user group forum:] Trying to see what panel file I have loaded.  Looked in Help->Context and it says "Current panel file: [none]  ".  Same whether I load a panel file file manually or even as a start-up action (that load does appear in the System Console as INFO from startup.PerformFileModel).  

I also get output of "None" when I enter this script:
import jmri
print jmri.configurexml.LoadXmlUserAction.getCurrentFile()

Shouldn't the loaded panel file name appear in both of these situations?

Thanks, Jerry

dsand47, 2020-12-30:
> [Issue copied from user group forum:] Trying to see what panel file I have loaded. Looked in Help->Context and it says "Current panel file: [none] ". Same whether I load a panel file file manually [...]

You can see the current xml data file name using **Panels => Show file history**.

In jmri.configurexml.LoadStoreBaseAction there are 3 static private JFileChoosers, one of which is userFileChooser.

When a xml data file has been loaded, the userFileChooser has the name that was loaded.  When the Store is invoked, the store process uses the **same** userFileChooser which already has the file name.  The file name only exists within the JFIleChooser object.

jerryg2003, 2020-12-31:
Thanks Dave.  Just trying to fully understand and hopefully make some helpful suggestions...

(1) The "Show file history" is interesting in that it shows the date and time my panel file was loaded, although not the version of JMRI under which it was loaded.  Might be nice to have this for debugging... (enhancement?).

(2) If I understand your point about userFileChooser, the name of the file loaded is private so not available to a script I might write.  Does that mean the script example I gave using configurexml.LoadXmlUserAction.getCurrentFile() is incorrect, or a bug in output?

(3) And, it still looks like a "bug" that the file name doesn't show up in the Help->Context.  

(4) I think it would be nice to have an "INFO" message in the console when a panel file is loaded, but I guess that becomes more of an "enhancement" issue. I have put INFO messages like that into all of my scripts so I can easily see what I've got loaded but the script and its message is, of course, under my control.

Thanks, Jerry

dheap, 2020-12-31:
@jerryg2003
A problem is that once a panel file has been loaded into JMRI its individual identity is lost. Its contents are merged with whatever already exists or has been loaded into the current session of JMRI.

So thinking in terms of the "currently loaded panel file" is incorrect and further reinforces the misconception that JMRI treats files like a word processor or spreadsheet.

A session of JMRI is like a mixing bowl into which you pour ingredients. Once the milk or eggs are poured in their identity is lost, you can't separate them out again.

Showing a "file history" for a session (what you poured in) may be a helpful concept.

dsand47, 2021-03-05:
PR #9482 includes a change to display the file name on the system console when loading a tables and panels xml file.

closed 2021-04-23 by dsand47

jerryg2003, 2022-04-11:
Resolved