JMRI/JMRI #9268: Pi Engineering Rail Driver Not Working

Spankyty, 2020-12-31:
Hello! New here so please bear with me, I have a Raspberry Pi 4 2GB running Linux, the premade image from here: [https://mstevetodd.com/jmri-raspberrypi-access-point](url)

I made the file 99-myhid.rules by doing these steps...
```
cd /etc/udev/rules.d
touch 99-myhid.rules
```

Then I did:
`sudo nano 99-myhid.rules`

Edited that file to exactly this:
      ```
  # My HID device
        ATTRS{idProduct}=="05f3", ATTRS{idVendor}=="00d2", MODE="0660", GROUP="plugdev"
```

ctrl+X, y, Enter

The group plugdev was already made and the user pi was already in that group.

I shut down panel pro, unplug the USB, wait a couple of seconds, replug the USB in, and open panelPro.
I click debug and RailDriver. RailDriver does not open anything. Nothing happens as far as I can tell. RailDriver (built-in) opens but it does not open saying RailDriver at the top. It acts like a normal throttle. RailDriver does nothing.

I do not have access to the internet with the Pi.

danielb987, 2021-01-10:
@Spankyty 
Welcome to JMRI !

I recommend that you ask on the JMRI Users Groups.io Group:
Join: https://groups.io/g/jmriusers/join
Browse: https://groups.io/g/jmriusers/topics

Spankyty, 2021-01-11:
> @Spankyty
> Welcome to JMRI !
> 
> I recommend that you ask on the JMRI Users Groups.io Group:
> Join: https://groups.io/g/jmriusers/join
> Browse: https://groups.io/g/jmriusers/topics

I'll do that.
Thank you!

closed 2021-04-14 by stale[bot]