JMRI/JMRI #9275: Support for DecoderPro Servo Function Output on ESU Loksound 4

richardhendo, 2021-01-05:
Hello,
I am using the recently added 'Servo' Aux Output function on the Loksound V5 L DCC decoder.
New support in DecoderPro works perfectly.
Now I want to put a servo on a Loksound V4 L decoder. The decoder supports 2 servos (at least).
But 'servo' is not supported on V4 on DecoderPro.
It seems the code for the V5 does the trick, can it be ported to the V4.
(Even though V4 is no longer sold, it is still around.)
Thanks, Richard.

dheap, 2021-01-05:
@richardhendo

I'll take a look at this as soon as I get a chance (between storms and domestic duties.

richardhendo, 2021-01-05:
Fantastic, no great hurry! 👍👍

dheap, 2021-01-06:
@richardhendo It turned out to be a trivial XML fix. I'll raise a PR soon and it will flow through into the next release

richardhendo, 2021-01-06:
O fantastic! XML is not trivial for me. My last codig was C+, before that COBOL!That would be 4.21.4? Will keep my eye out for it. And buy another servo...Thanks,Richard.

dheap, 2021-03-30:
Fixed by #9277

closed 2021-03-30 by dheap