JMRI/JMRI #9295: Create Test Release 4.23.1

danielb987, 2021-01-13:
This is the next release in the 4.24 cycle. It's intended to be created from the HEAD of the master branch.

bobjacobsen, 2021-03-06:
The release-4.23.1 branch has been created. 

Maintainers, please set the 4.23.23 (note!!) milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](https://builds.jmri.org/jenkins/job/testreleases/job/4.23.1/)

silverailscolo, 2021-03-07:
Milestone 4.23.3 (dot three) is next, with 4.23.2 reserved for LogixNG, correct?

danielb987, 2021-03-07:
> Links:

Yes.

Daniel

2021-03-07 15:03 skrev Egbert Broerse:

bobjacobsen, 2021-03-08:
JMRI 4.23.1 has been released. Files are available in the GitHub release section.

closed 2021-03-08 by bobjacobsen