JMRI/JMRI #9298: how to find back my previous bug description ?

petermill, 2021-01-14:
Issue Type: __Bug__

Sorry but I am new to the site and process.

I left a bug description about Programming om Main and I can't find it back on GitHub.
I wrongly detected  an error in the code for POM read.
e6 30 00 07 e4 00 00 35
Actually e6 30 with code E4 after the CV is  POM read for version V3.6

I need to check back the OpenDCC itself

PanelPro 4.21.3+R63094d48b
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 13.0.2+8 mixed mode, sharing
Windows 10 10.0 amd64

danielb987, 2021-01-14:
That issue is here: https://github.com/JMRI/JMRI/issues/9296

danielb987, 2021-01-16:
I'm closing this issue now. Feel free to reopen any of your issues or create a new one if you need more help.

closed 2021-01-16 by danielb987