JMRI/JMRI #9315: Missing directory in the JMRI website

dsand47, 2021-01-21:
The scripts directory is missing in https://www.jmri.org/help/en/html/.

[image]

[image]

dsand47, 2021-01-30:
@jerryg2003 
Do you have any way to fix this?

jerryg2003, 2021-01-30:
Hmm... don’t know why that would be. In Git, the scripts directory is there but has no files and has only a single sub directory, yaat. Perhaps if you put some file in scripts, such as an index.shtml, that would cause the next update to create that part of the tree on the web. I don’t know how JMRI processing decides when to create new directories on the web so unfortunately this is only a swag.
[I note that the doc directory is similar in that no files and a single sub directory and it exists on the web so perhaps my guess is off base, but that’s all I can think of].

dsand47, 2021-01-30:
On JMRI/JMRI it looks strange. but OK.

[image]

[image]

If you look at the Apache index listing in the first post, doc is ok but scripts is missing.

I think doc and scripts look strange because they only have one entry and GitHub is condensing the path.

Someone with access to the Apache server will need to look at this.

mattharris, 2021-01-30:
I'll take a look over the next few days.

bobjacobsen, 2021-01-30:
I expect the the problem is the "exclude-list" file.

There's a Jenkins task that uses rsync to move files from a local (to Jenkins) checkout to the Apache web server's file system.  The command is:

```
rsync -avz -e ssh -i /var/lib/jenkins/.ssh/jmri_id_rsa -l jmri --times --exclude-from=exclude-list --include-from=include-list /var/lib/jenkins/workspace/website/jmri-repository/ jmri.emjay.info:/home/jmri/htdocs
```

The exclude-list file contains the line:
```
scripts
```
which is might be excluding a directory called "scripts" anywhere in the path, not just at the head.

mattharris, 2021-01-30:
I think you're right as I believe that rsync excludes are interpreted as relative.

dsand47, 2021-01-30:
I could change the directory to something like scripthelp or scripting and update the JavaDoc which is the source for the toc and index.

bobjacobsen, 2021-01-30:
Hopefully #9362 will fix this once merged and Jenkins recycles.

jerryg2003, 2021-01-30:
Dave -

I actually like the idea of changing it to “scripthelp” so people don’t just start posting scripts there rather than to the jython directory where all others reside.  As I recall, you created this directory so you could put in some detailed instructions that didn’t seem like they should go in the jython directory.

Jerry

________________________________
JerryG2003@aol.com

dsand47, 2021-01-31:
> Dave - I actually like the idea of changing it to “scripthelp” so people don’t just start posting scripts there rather than to the jython directory where all others reside. As I recall, you created [...]

@jerryg2003 I agree about changing the name.  PR #9366 has been submitted.

I don't expect see a lot of detail help pages for scripts but the location will be available for future help files.

dsand47, 2021-02-01:
This has been fixed by PR #9362.

closed 2021-02-01 by dsand47