JMRI/JMRI #9317: Dispatcher Auto train at continuos running

Shegar81, 2021-01-21:
I have a simple layout composed by an oval with two stations that have either two tracks. A simple layout with detection on all tracks. Each track of the station is divided on ther sections. Between turnouts there is three sections with current detection. All is with feedback. Also the turnout has feedback by two sensors. Hardware is from MERG group the DCC components by CBUS. 

With this layout I am running trains trying to cross them with automatic trains. I assign the train with the dispatcher a transit without problem. At the end of the train it didn't stops but finally I got it. I put an stopping sensor on the last section and it worked quite well. When I am running on continuous running is when I have some problems. I detected that at the last allocation train didn't get the last signal mast so it doesn't take care about it. So if I block the allocation of the first step on the second loop for the train it doesn't take care about the mast and even if it is on danger it goes up. I detected two sitautions. First if the last order is to pause after the pause train continues with speed of last order even if the turnout is on oposite position. Second if the train stops by stopping sensor the train doesn't restart when the allocation is possible.

Pugwash1, 2021-01-22:
@Shegar81 
I would need to see the Layout Panel XML and how you are setting up trains to debug it.
If you dont have block lengths then the train will stop when entering final section, if you have block lengths & train length and the train fits then it will stop when penultimate block clears. The blocks must go inactive in the order they went active, If using stoppingbyspeed profile, the profile must be acurate and the block lengths must be accurate. The correct scale must be set in 2 places, Warrents (in prefernces) and in Dispatcher options.
When you try your first train, have only that train active on the layout, and do a real simple start from section one go to section two, then expand by one section at a time. If using automatic generated sections you cannot go "back and forth" if a switch is involved. If using automatic generated sections you will need to add Direction Sensors to the SML as the SML and Sections generate a "Permissive" system, the direction sensors make it an absolute system.
If you are running autotrains, then all trains, including manual trains, must be dispatched, else the program will loose track of where the trains are.

Shegar81, 2021-01-23:
I have the length on the blocks and it is bigger than the trains. 
I did a signalling system trying to works as the Spanish railway. Signalmast aspects are defined and the way I make that a Signal Mast use one or other aspect is by logix. 
How can I send you the layout panel? I think it is quite long to show it here.

Pugwash1, 2021-01-23:
@Shegar81 
Before trying to automate, or even signal a large layout, practice with a very very small part of it.
Start with 2 blocks, 2 sensors, 2 sections, 1 transit, Virtual Signals, "Basic Model Signals" (red,abmer,green-stop slow,normal) and make sure you can get that to work. Then add a switch and another section of track, and make that work, until you have confidence the system works and you can make it work. Then see if you can make a custom signal system work. 
Only after all that start to apply that knowledge to the layout, checking each bit as you go. Its very easy to debug a transit when it stopped working when a new section was added, but very difficult to debug if you added 10 sections and twenty signals and used a custom signal system (or even a "prototypical" signal system.

Shegar81, 2021-01-23:
Hi,

This is the idea with the layout I have. I am testing. It is a layout of 4 tournouts on a oval trying to simulate two stations of two platforms each one. I did what you recommended yet. I have some fails and I learn about these errors. An only train in automatic mode works fine. But when I put the second one with interaction about this management of the free track it is when I had the problem. I had some problems about the train stops and I realise that the automatic train was using a signal that was not on the blocks it was working. This problem I assumed it was possible because some test I was doing. The issue I reported I check it with diferent points of the layout, diferent transits and diferent situations. I realised it finishing the transit with a pause or without it. 
I insert the file on a zip to share here. It is a layout of 4040 lines. 
[Nou blocks4.zip](https://github.com/JMRI/JMRI/files/5860297/Nou.blocks4.zip)

Shegar81, 2021-01-23:
I have to add that I use normally going from safe position to safe position on the automatic trains

Pugwash1, 2021-01-23:
@Shegar81
You didnt include your custom signal system. I replaced all the custom masts with Virtual, Model Railraod, Two lows, rebuilt the logic. You have direction sensors on the section but not added to the SML logic. The logix you have appear to be making 3 blocks into 1 block, I dont know why. Trains appear to run correctly when using the block sensors on and off and stop without stopping sensors. 
I suspect that either the signal system or the logic has errors. The signals must be red/stop/ zero speed when the track ahead is not allocated, this is done by including direction sensors in the SML.
Try uiing the virtual masts and no logic and make sure that works, before attempting to go futher.

Shegar81, 2021-01-23:
[RFIG.zip](https://github.com/JMRI/JMRI/files/5861023/RFIG.zip)
I add here the signalling. To get signals clear only if there is an allocated section you have to have active the "button" or sensor Dispatcher on. The 3 blocks on one is to put on the block the stoping sensors. 
About the continous running I have a positive action putting a delay. If I introduce the delay at the end before the next round starts 1-2 fast minutes it works fine. Only I have a problem if I end with a pause. If the last order is a pause after the pause the train starts even if the is no allocation. It starts besides it is on the waiting mode after the paused.

Shegar81, 2021-01-23:
@Pugwash1 Thanks for your time and help

Pugwash1, 2021-01-24:
@Shegar81 
Do not end with a pause. There is no point, and will likely mess things up, as it will try and restart a train thats paused, not a train that has reached the end of its transit. To Pause between runs use "restart when done", and add delay, or sensor if you want to control the restart time.
I will look at the signal system, but its unlikely I can help with that. Providing your signal system says stop, and stop means stop, when track ahead is not allocated (direction sensors correctly set and processed) it will at least stop when it should.

Shegar81, 2021-01-25:
Thanks for your help.
I realise that as you said it is better to use the delay than the pause at the end of transit. I checked that it was not the last version of JMRI so I was on 4.18 , now it is updated yet. After checking logix and all it seems to worrk fine. Now my request is more on the autoallocate system as I have some crashes that doesn't use correctly the alternate when there is an ocuppied block. After all tests I read that multiple trains are not well working, so if I can help on that I want. I am not on programming but i have some knowledge about it and I can learn.

Pugwash1, 2021-01-26:
@Shegar81 
The issues I know of with alternates are that you cannot start or end on an alternative, and if you have a transit that goes round a loop a couple of times then if the starting block is later used as an alternative it will always choose the starting block (it applies the rule about cannot start or end on an altenative, I keep meaning to fix that but never quite get around to it)

Shegar81, 2021-01-27:
@Pugwash1 Thanks for the information. I try to respect the idea you put. On the autoallocation I found that for my layout if I want to move trains on both directions at same time it is necessary that there is a free platform on each station, if it is only to move all trains on same direction is with only one free platform.
As you said to use the alternate options the transit must be only with one loop, can't have until now more than that and without pause at the end.
As I said if I can help you with this part I will be very glad.

Shegar81, 2021-01-29:
Finally most of the traisn goes perfectly. I found that the block speed of the block is not respeccted. On auto train the block restriction is not considered. It runs as the signal mast indicates or action requires. If it is next to the stop also it reduces correctly. I solved the restriction because of allocation for 3 trains by pause allocation. 
Using the back and forth running any option or trick to do actions work on one direction or the other?

Shegar81, 2021-02-06:
I found the problem about the blocks. As the signals were working with the logix instead of SML it was not considering the block limitation. When I did a better implementation of my signalling system and I am working with the SML it started to work fine.

closed 2021-04-14 by stale[bot]