JMRI/JMRI #9323: Updates to Decoder Definition file Bachmann_EZDCC.xml

klausnahr, 2021-01-23:
Hi,

I did some updates to the Bachmann Decoder Definition file Bachmann_EZDCC.xml:

- added Decoder N Scale EMD F7A
- added Decoder N Scale EMD F7B
- added manufacturer infos
- updated Accel/Decel limits
- updated CV50 default
- added function Dim
- updated function mapping
- changed reset to 8
- hint on pane Bachmann
- fixed non working tooltips
- added German translations

Best regards
Klaus Nahr

[Bachmann_EZDCC.zip](https://github.com/JMRI/JMRI/files/5861024/Bachmann_EZDCC.zip)

dheap, 2021-03-30:
User has submitted update files but these haven't been converted to a Pull Request.

klausnahr, 2021-04-09:
I would love if somebody else could do a pull request as I still have no knowledge about how to do that.

Klaus

Re465, 2021-04-09:
I'll do it tomorrow or On Sunday

closed 2021-06-08 by stale[bot]