JMRI/JMRI #9324: New Decoder Definition File for ANE Model

klausnahr, 2021-01-23:
Hi,

I created a new Decoder Definition file for ANE Model decoders from the scratch. Currently there is only the basic definition file ANE_LC201.xml shipped with JMRI. This lists only LC201 and has no nice function mapping.

The new file includes LocoCruiser:

- LC101
- LC102
- LC201
- LC202
- LC203
- LC204
- LC205
- LC206
- LC301
- LC302
- LC303
- DL28

Best regards
Klaus Nahr

[ANE_Model_LocoCruiser_V201.zip](https://github.com/JMRI/JMRI/files/5861034/ANE_Model_LocoCruiser_V201.zip)

Re465, 2021-01-24:
LC101 and LC102 have value 1 in CV7 and a number of other CVs you have to use exclude with these decoders.

Wouter

klausnahr, 2021-01-24:
I did the definition file based on the documentation published by ANE Model. For LC101/102 I did only an exclude for CV10.

From LC302 I know that not all documented features really work. I already had some discussion with ANE Model about that. They told me, all features should work. Maybe they have different Firmware with same version number. So I decided to include all features in the Decoder Definition file as reading and writing of the documented CVs is possible even if the actual value of some CVs ignored.

Klaus

Re465, 2021-01-24:
I see big differences.
LC101 and LC102 uses CV9 and no CV10 and 11
CV33-38 is at LC101 and LC102 CV55-59.
CV49-54 is at LC101 and LC102 CV60-64.
And I can't find CV41 (Brake) in the user manual.
User manuals consulted: LC101 and LC203 From this side.
https://www.anemodel.com/DL1.html.

klausnahr, 2021-01-24:
The manuals for LC101 and LC102 a very old and probably outdated, they reference to version 1.1. I was told the decoders are now version 2.3 too like the others, only no BEMF. New version of the Basic series should actually already available, at least I was told a few month ago. Also there are some small differences between english and chinese manuals and some bugs in the manual too. I wrote a german manual for the Decoders and spend a lot of time consulting all available manuals and get some feedback from ANE Model too. CV41 is documented in the _LC201 Lococruiser HO standard decoder _with 8 pin plug_ 6 output user manual.pdf_ and is working (at least when not in advanced consist mode).

Klaus

Re465, 2021-01-24:
Ok, that doesn't speak for this manufacturer that se don't have the latest manual on the homepage.

Then I would put behind LC101 and LC 102 (Version 2.3) so also done be Zimo so that if a template for the version 1.1 is created there is a different.

Will I make a pull requests for the template or do so yourself?

Wouter

PS where can I download the manual for version 2.3.

klausnahr, 2021-01-25:
Actually I do not really understand what you mean with "Will I make a pull requests for the template or do so yourself?".

The 2.3 Manual is available under the link which you posted before.

This is what ANE Model wrote to me:
- LC2xx series DCC decoder firmware is the same as LC3xx.
- We have a mistake in the function output define in the LC302 use manual. It should be the same as LC201.
- Our DL-28 DCC decoder is the same as LC201 and LC302.

I am not sure if the final DL-28 decoder will have Analog mode. At least I hope so, they currently working on a new release.

I will check with ANE Model about the LC101/102 firmware. The manual doesn't even list long addresses.

Klaus

danielb987, 2021-01-26:
> Actually I do not really understand what you mean with "Will I make a pull requests for the template or do so yourself?".

A "Pull request" is a request to the maintainers of JMRI to include your changes into the JMRI code base. Since @Re465 offered to do that for you, you can let him do it if you want. We appreciate your work and is here to help you.

But the process of adding new stuff to JMRI is:
- Fork the JMRI/JMRI repository to your GitHub account
- Clone the klausnahr/JMRI repository to your local computer using a git client
- Create a new branch for your changes
- Do the desired changes, like adding a new decoder definition
- Commit your changes to your branch
- Create a Pull Request (PR) for your branch
- The maintainers of JMRI will then review your work and if everything is OK, we will merge your changes into the JMRI code base.

If you are using Windows, it's recommended to use GitHub Desktop for this. See this document:
https://www.jmri.org/help/en/html/doc/Technical/githubdesktopinto.shtml

klausnahr, 2021-01-26:
OK, I understand. But let's wait until I get feedback from ANE Model about the features / version of the current LC101 / LC102 decoders.

Klaus

klausnahr, 2021-04-09:
I got a confirmation from ANE Model that my previous information about the LC101 and LC102 was wrong. These Decoders are also no longer available.

As the feature set of Version 1 (LC101, LC102) and Version 2.01 (LC201, LC202, LC203, LC204, LC205, LC206, LC301, LC302, LC303, DL28) is totally different I have build two Decoder Definition Files.

I would love if somebody else could do a pull request as I have still no knowledge about that. Same for my update of the Bachmann Decoder File #9323.

Klaus

[ANE_Model_LocoCruiser_V201 2021-02-09.zip](https://github.com/JMRI/JMRI/files/6285337/ANE_Model_LocoCruiser_V201.2021-02-09.zip)
[ANE_Model_LocoCruiser_V1 2021-02-09.zip](https://github.com/JMRI/JMRI/files/6285338/ANE_Model_LocoCruiser_V1.2021-02-09.zip)

danielb987, 2021-04-09:
@Re465 
Can you do the pull request for @klausnahr ?

> [ANE_Model_LocoCruiser_V201 2021-02-09.zip](https://github.com/JMRI/JMRI/files/6285337/ANE_Model_LocoCruiser_V201.2021-02-09.zip)
> [ANE_Model_LocoCruiser_V1 2021-02-09.zip](https://github.com/JMRI/JMRI/files/6285338/ANE_Model_LocoCruiser_V1.2021-02-09.zip)

Re465, 2021-04-09:
Yes I can make the pull requests

klausnahr, 2021-04-09:
@Re465 Thanks a lot! Could you do it for #9323 too?

klausnahr, 2021-04-09:
I forgot to mention, the old Decoder File ANE_LC201.xml is now obsolet and should be removed.

bobjacobsen, 2021-04-09:
There's a discussion of how to replace an older definition with a newer one here:  https://www.jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml

There are some migration steps because people will already be using the old one.

klausnahr, 2021-04-09:
OK. I have overseen this. I will update ANE_LC201.xml.

klausnahr, 2021-04-09:
@bobjacobsen @Re465 I have updated the old Decoder Definition File ANE_LC201.xml and successfully tested the migration via "Update Decoder Definitions" to the new one. Updated file attached. Thanks for your help.

Klaus

[ANE_LC201.zip](https://github.com/JMRI/JMRI/files/6288713/ANE_LC201.zip)

Re465, 2021-04-10:
Before I made the pull request I have already adjusted "ANE_Model_LocoCruiser_V201.xml".

closed 2021-04-10 by danielb987