JMRI/JMRI #9327: Switchboard Loop

dsand47, 2021-01-23:
When the **Switch shape** is other than **Button**, the Java process loops.  It is still possible to click on the panel.

JMRI 4.21.3, Java JDK 1.8, macOS 10.15.7.

[SB_Loop_Test.txt](https://github.com/JMRI/JMRI/files/5861298/SB_Loop_Test.txt)

silverailscolo, 2021-01-24:
Thanks for reporting. How do you see the loop? Flickering on screen or repeating lines in the JMRI System Console?

dsand47, 2021-01-24:
> Thanks for reporting. How do you see the loop? Flickering on screen or repeating lines in the JMRI System Console?

The CPU utilization jumps to about 140%.  There is no apparent screen activity and no console messages.  I tried enabling debug logging for the SwitchboardEditor package.  The log showed what I assume are typical debug messages.  If I let it sit, the CPU keeps going without any debug messages. 

During the looping, it is still possible to click on switchboard icons and change sensor states and the switchboard icon changes.

silverailscolo, 2021-01-24:
By adding a log statement in the BeanSwitch `paintComponent` method I confirm it was called all the time. After searching around, a last (!) search hightlighted a `repaint()` call I added in Oct. 2020, creating a 'perfect' loop.
Apologies to anyone affected - or just bothered - by this bug. Fixed in PR #9326 (for 4.23.1).

dsand47, 2021-01-25:
I have tested PR #9326 and the looping does not occur.  Thanks.

danielb987, 2021-01-26:
#9326 is now merged. Can this issue be closed?

dsand47, 2021-01-26:
Yes.

closed 2021-01-26 by danielb987