JMRI/JMRI #9357: 1Sensor Turnout Feedback unchanged when Sensor goes unknown

icklesteve, 2021-01-30:
With 1 Sensor Turnout feedback, when the Sensor being followed changes to UNKNOWN or INCONSISTENT , the Turnout feedback does not update.

When 2Sensor feedback is used the Turnout feedback is updated correctly.

KenC57, 2021-01-30:
I would say (ONESENSOR) that if the turnout went to unknown, following the
sensor, that would make sense. It shouldn't stay at CLOSED or THROWN.

For a TWOSENSOR, either of those sensors being other than ACTIVE or
INACTIVE, I would say being UNKNOWN or INCONSISTENT would fit. Leaving it as
CLOSED or THROWN would not be correct.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.org
www.syracusemodelrr.org

icklesteve, 2021-01-30:
Hi @KenC57 

I've added 2 tables with the possible combinations of all Sensor status for both 1 Sensor and 2 Sensor Feedback to #9358 
1 Sensor feedback Unknown / Inconsistent > Inconsistent Turnout is the only program change.
2 Sensor feedback all unchanged, just documented which state they enter given the inputs, eg INCONSISTENT && UNKNOWN = INCONSISTENT.

PR #9358 (2021-02-07): Turnout Feedback - Improve 1 & 2 Sensor / Support Page
Updates 1 & 2 Sensor Turnout Feedback Support Page.

Adds support for 1 Sensor Feedback UNKNOWN and INCONSISTENT states ( fixes #9357  ) .
closed 2021-02-07 by danielb987