JMRI/JMRI #9361: Function Panel empty with all locos

Glasaudio, 2021-01-30:
Just start to juse JMRI, and after a some days the Function Panel is empty. I cant switch the funtions, but the throttle works.
I have the latest version instaled.

[image]
[image]

dheap, 2021-01-30:
@Glasauidio
Open Preferences->Throttle and find the setting that shows or hides unused Function Keys.

Change that setting so unused keys show.

If you have it the wrong way, you won't see any Function Keys unless you've entered custom names in Labels and Media for every loco.

dheap, 2021-01-30:
@Glasaudio
Sorry, I'm not at a commuter at present to check exact wording of preference.

Glasaudio, 2021-01-30:
Thanks for the reply. I have found of, and it Works. Thanks jou

Gideon

Outlook for Android<https://aka.ms/ghei36> downloaden

silverailscolo, 2021-01-30:
Hi Gideon, welcome to JMRI and good to see you received enough advice to use the throttle (and glad to see you are using the Dutch locale; veel plezier met JMRI, stap voor stap werkt het best).
Closing this issue.
Egbert

closed 2021-01-30 by silverailscolo