JMRI/JMRI #9369: CI Cucumber - Firefox webserver fails

silverailscolo, 2021-01-31:
Repeated >2 weeks webserver/Web Table failure on Jenkins Cucumber:
``

Failed

JMRI Web Table.Web Table requests (from cucumber.runtime.formatter.JUnitFormatter)

Gefaald voor de laatste 6 bouwpogingen (sinds' 'Gefaald#94 )
Duurde 31 ms.
Error Message

java.lang.NoClassDefFoundError: org/apache/hc/core5/http/ClassicHttpRequest
	at jmri.util.web.BrowserFactory.getBrowser(BrowserFactory.java:44)
	at jmri.web.WebServerAcceptanceSteps.lambda$new$0(WebServerAcceptanceSteps.java:48)
	at ✽.I am using firefox(file:java/acceptancetest/features/web/WebTable.feature:5)
Caused by: java.lang.ClassNotFoundException: org.apache.hc.core5.http.ClassicHttpRequest
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
(...)
``
See https://builds.jmri.org/jenkins/job/development/job/cucumber/99/testReport/junit/(root)/JMRI%20Web%20Table/Web_Table_requests/

mstevetodd, 2021-02-03:
@silverailscolo What is this test for? This same test is run as part of the Travis CI, isn't it? Except that one works.
I recently (PR #9279) enabled the firefox tests (by removing the @Ignore), and verified they ran on Travis. I didn't realize there was this other test run out there.

silverailscolo, 2021-02-04:
Sorry @mstevetodd, I have no idea what the test exactly does, but I want to report here that the Jenkins Cucumber tests show a dark thundery cloud (see link above) and this test set has not successfully run since somewhere in November of last year.
Current report:
[image]
Details: 0 passed
[image]
The report repeatedly describe each test exited at some`classNotFound ... ClassicHttpRequest` whether using Firefox or Chrome.

mstevetodd, 2021-02-04:
OK, thanks.

My suggestion is to turn off the Cucumber test, since it isn't working, and no one is apparently using it. Those same tests are running successfully elsewhere.

Of course, I have no idea where that is controlled.....

closed 2021-04-14 by stale[bot]