JMRI/JMRI #9371: Sensor table fails after adding more than one sensor

GreyMark7650, 2021-01-31:
I have a system running on Raspberry Pi using DCC++ for control and turnouts and C/MRI for sensors. When adding sensors the first one will add correctly, but all after that fail to display although they may actually be created. The sensor item in the left menu disappears although it is still present in the tables menu from PanelPro.

Checking the .xml file that is saved I can see that the sensor entries after the first are truncated with only the sensor ID present. While I can edit the .xml to add new entries, loading the file still causes the menu items to disappear. This is true even if there is only one sensor in the file.

Once the file is loaded, the sensors are accessible and useable, for example in Switchboard, but the sensor page is missing from tables and so is impossible to edit.

dsand47, 2021-01-31:
Are there errors in the system console?

Help => System Console.  Use the **Copy to clipboard** button to copy the contents and paste them in a reply.

GreyMark7650, 2021-02-01:
Sorry for the delay, the computer is remote and not connected to the internet.

I can see an error in the console file when attempting to open the sensor table. The third file is the .xml loaded.

[console_log_1-2-21.txt](https://github.com/JMRI/JMRI/files/5903048/console_log_1-2-21.txt)
[log_after_table_open.txt](https://github.com/JMRI/JMRI/files/5903049/log_after_table_open.txt)

[tintern_sensors_turnouts_panels_xml.txt](https://github.com/JMRI/JMRI/files/5903056/tintern_sensors_turnouts_panels_xml.txt)

dsand47, 2021-02-01:
> Sorry for the delay, the computer is remote and not connected to the internet.
>
> I can see an error in the console file when attempting to open the sensor table. The third file is the .xml loaded. [...]

Edit `/home/pi/.jmri/Tintern_Railway.jmri/profile/c6f3cdb0-4677-11e9-8000-b827eb71f7dc/user-interface.xml`.  The random looking part might be different.

Remove any elements with the following name:  `<table xmlns="" name="jmri.jmrit.beantable.SensorTableAction:...">`

If that does not work, delete user-interface.xml.  You will lose window positions and other retained settings.  They will be re-created when PanelPro is started.

I suspect that the mixture of DCC++ and CMRI in the **All** sensor view results in some un-sortable columns.

GreyMark7650, 2021-02-01:
"Remove any elements with the following name: <table xmlns="" name="jmri.jmrit.beantable.SensorTableAction:...">"

Fixed the issue on my system. Thanks.

Is that a permanent fix or is it likely to return on update etc.?

MarkW

dsand47, 2021-02-01:
> "Remove any elements with the following name:
> 
> "
> Fixed the issue on my system. Thanks.
> 
> Is that a permanent fix or is it likely to return on update etc.?
> 
> MarkW

As long as you only sort on the main columns, such as system name, user name, comments, you should not have an issue again.  My guess is that there is a column that CMRI and DCC++ use for different data types, i.e. string vs. integer.

icklesteve, 2021-02-09:
Hi @GreyMark7650 

I've tried to reproduce this using simulated CMRI + DCC Connections, 
however with many combinations attempted I can't get it to fail.

Do you happen to have kept the offending contents of <table xmlns="" name="jmri.jmrit.beantable.SensorTableAction:..."> ?

Ta, Steve.

GreyMark7650, 2021-02-09:
Hi Steve,  unfortunately not. Sorry. 

⁣Get BlueMail for Android ​

icklesteve, 2021-02-09:
No problem, if it re-ocurrs you know what I'm looking for :-)

closed 2021-02-24 by silverailscolo