JMRI/JMRI #9411: Update for Zimo "Swiss Mapping"

nigelcliffe, 2021-02-14:
Attached ZIP file contains four files to update Zimo decoders. 

Three files replace current JMRI files in       xml/decoders/zimo   
One file is new, also destined for                xml/decoders/zimo

This adds features currently absent from the Swiss Mapping panes, specifically to select the dimming group for each output in the table,, and to set the dimming levels.  

Can someone please do the steps to incorporate these in releases.  Thanks.  

[swissmappingUpdate.zip](https://github.com/JMRI/JMRI/files/5977638/swissmappingUpdate.zip)

Nigel

silverailscolo, 2021-02-23:
Will do tonight, thanks for the contrib, @nigelcliffe

nigelcliffe, 2021-02-23:
Thank you,  @silverailscolo 

I'm still considering how to deal with bits 5&6 of the m-key field.  That might lead to a further update if I can come up with a sensible solution.

silverailscolo, 2021-02-23:
will close when merged

closed 2021-03-03 by silverailscolo