JMRI/JMRI #9412: ESU causes Heap Error - When Opening Comprehensive Programmer

WSampsonite, 2021-02-14:
Any locomotive, that has a ESU LokSound V5, will error out, with a memory issue, stating there’s a “Heap Error”.

It has been narrowed down to the V5 Decoder, since the ESU Select and earlier ESU decoder versions load fine (along with WOW and Tsunami).

I was attempting to add a new loco, it’ll read the locomotive just fine, it’s ID the V5...as soon as I click “Open Comprehensive Programmer”, it’ll spin, then error out after 2-3 minutes. The same goes, if I click on any of the other locomotives, with V5’s installed.

I’ve tried updating to the latest JMRI and even tried out the Latest Beta. Uninstalled both completely, along with Java. Took everything off and reinstalled each.

I’m more than willing to share video or show the error happen. I can make it do it on command. So don’t hesitate to contact me directly.

[image]
[image]
[image]
[image]

dheap, 2021-02-14:
@WSampsonite There's no doubt ESU LokSound 5 definitions require a fair bit of memory. The 5 definition needs over 2,000 CVs and nearly 10,000 Variables (individual settings). That's over double the numbers in a V4.

However, on my Train Room Mac, an old machine with only 6GB of RAM, I was able to successfully have 4 simultaneous LokSound 5 definitions open at once, although they became successively slower to open. It seems to have lost the plot when I tried to to open the fifth, but no heap memory error yet...

Java VisualVM tells me the JMRI startup script has allocated 1.5GB of heap space (-Xmx1536m) on this particular old machine.

I'm not sure how you determine that allocation with Windows startup. Someone else more familiar with that area will be able to tell us.

Meanwhile, you could upload the results of the Heap->Context menu item.

Dave

bobjacobsen, 2021-02-14:
In PanelPro, Debug > Memory Usage Monitor and click “Update” to see immediate used and free heap memory.

I don’t know how to do that under DecoderPro, unfortunately.

Bob

> On Feb 14, 2021, at 1:36 PM, Dave Heap <notifications@github.com> wrote:
> 
> I'm not sure how you determine that allocation with Windows startup. Someone else more familiar with that area will be able to tell us.
> 
> 

—
Bob Jacobsen
rgj1927@gmail.com

dheap, 2021-02-15:
> @bobjacobsen In PanelPro, Debug > Memory Usage Monitor and click “Update” to see immediate used and free heap memory. I don’t know how to do that under DecoderPro, unfortunately. Bob

- Open DecoderPro and go to the File Menu.
- Pick "Open PanelPro Window".
- With the PanelPro window foremost, you'll find the Debug Menu -> Memory Usage Monitor.
- You'll see total memory displayed straight away.
- Click “Update” to see used and free heap memory.
- Leave the Memory Usage Monitor and PanelPro windows open and click back on the main DecoderPro window if you want to open roster entries, etc.

dheap, 2021-02-15:
@WSampsonite
- With the PanelPro window foremost, you'll find the Debug Menu -> Memory Usage Monitor.
- You'll see "total memory" displayed straight away.

It's the "total memory" figure we need to know. Get that as soon as you start DecoderPro, before trying anything.

I'm suspecting that for some reason, LaunchJMRI.exe isn't allocating enough Heap Memory on your machine.

The results of the Help->Context menu item will also assist us to see what your particular environment is.

dheap, 2021-04-01:
Unable to pursue this issue unless @WSampsonite (the original poster) provides the requested information.

WSampsonite, 2021-04-01:
We never found any resolve, but have opted to program ESU Version 5 locos
on the LOK Programmer.

dheap, 2021-04-01:
@WSampsonite
Unfortunately your failure to provide the requested Information means we can't pursue any possible memory allocation problem in our code with a Windows machine having a similar memory size.

WSampsonite, 2021-04-01:
I do not fully understand the request. We’re managing to get things to
work, for now. If any issues become unbearable, we’ll reach out and revisit
the issue, with video to show what’s happening.

Sorry we didn’t get the protocols needed to move forward. I’m not familiar
with the request process, but really appreciate the info previously
provided.

danielb987, 2021-05-17:
@WSampsonite 
I'm closing this issue now. If the problem happens again, you are welcome to open the issue or to create a new issue.

closed 2021-05-17 by danielb987