JMRI/JMRI #9419: Rendering of PDF files in JMRI General Help

jerryg2003, 2021-02-17:
Is it true that PDF files cannot be displayed in the JMRI Help?  There has been some discussion on the user forum about people wanting to contribute information in PDF files.  BobJ has advised against that in the user forum, but I have another request from a person developing Operations documentation as PDFs.

Examples:
I tried clicking on "Manual" and get nothing; all of the manuals are PDFs.
I tried clicking through OpenLCB ->  OpenLCB Signal Masts -> "technical summary on masts" (which is a PDF file) and got uninterpretable characters.

Also, is it possible to list TOC and Index entries that point to PDF files by appropriate entry in Map.jhm?  So far, there are none.

If the system can't display PDF files in JMRI General Help, then we probably should have an alert that people may need to go to the web site for certain content already.

[This may be related to an old discussion in Issue #6497 about changing the Help system completely, but that appears to be a dormant topic.]

Thanks, Jerry

bobjacobsen, 2021-02-17:
Several things:

1) There’s no problem contributing PDF files _as_ _long_ _as_ the original source file is _also_ contributed.  Word, LaTex, whatever, without that original file the only way to update or maintain the PDF is to throw it out.  So it hangs around with obsolete info in it.  That’s a concern, right?

2) JMRI uses JavaHelp to display the help within a JMRI window (https://www.jmri.org/help/en/html/doc/Technical/Help.shtml)  It can display HTML4 from both local files. If it doesn’t know how to display a file, it invokes the default browser to display it.  For example, open help, if need be go to “JMRI Help Main Page”, click on the link the main JMRI web page just below “Further Information”.  That’s not a link to a local file, so it (should) be opened in your default web browser.   The  OpenLCB -> OpenLCB Signal Masts -> "technical summary on masts” mentioned below works for me. But perhaps there’s a problem with some particular machines with invoking the browser.  

3) I’m not sure about the point about “Clicking on “Manual””. I’m probably looking in the wrong place, but when I click Manual_s_ I get a list of the manuals, most of which are HTML format.  There’s one noticeable PDF file, which displays fine in my default browser.

What’s the right way to see the problem?

Bob

> Is it true that PDF files cannot be displayed in the JMRI Help? There has been some discussion on the user forum about people wanting to contribute information in PDF files. BobJ has advised against [...]

—
Bob Jacobsen
rgj1927@gmail.com

bobjacobsen, 2021-02-17:
The image that got lost in the email processing above:
[image]

jerryg2003, 2021-02-17:
That's what I was doing - on Windows 10, using 21.4, Chrome browser.  Can't seem to paste my screen shots in here so will email you.

bobjacobsen, 2021-02-17:
PR #9422 should improve this on Windows. I don't have a way to test on Windows 10, unfortunately.

silverailscolo, 2021-02-20:
To get back to your initial question:
Thanks to Bob's recent PR the user can more easily access PDF's from the help, but these documents will jump out of the JHelp window. PDF's are page based so they are not well suited to replace help of a particular JMRI window, fully integrated in the JHelp viewer. For example, when there's just 5 lines of text, a PDF will still present that on a Legal or A4 sheet. Portrait or Landscape, another relevant thing to be clear about.
PDFs are easy to produce so it logical users suggest to contribute, but unless they follow the same rules as the help pages, they are flat and not structured the way information on the web is (supposed to be), allowing access on different screens, reflow content and use in text-to-speech.
As Bob has stated, adding multi-page "manuals" in PDF format linked to the help (site), that's great.
Instead of the tree-like structure of the code, it allows telling the user case story.
To make that clear, I suggest we only support a few "bins" to store them (Website repo under Manuals?), add useful hyperlinks in the JMRI Help, and also the other way around!
And let's make sure ahead of time that the "source" of the PDF is shared as well.
Also mind the size of these PDF's. Another reason to put them in a specific place, so GitHub 'pulls' stay manageable.
Egbert

silverailscolo, 2021-02-22:
I read some of the examples of PDF how-to sets as shared in the JMRI-users group. For their actual contents they have little to do with the Pdf format and could better be inserted as concise web pages. Printing those will get you a pdf without any frills, effectively the pdf now being shared. But its contents will be better indexed and turn up in searches.
We drop them in a folder using short simple names, integrate in GitHub and link to them from an index web page, presto.

jerryg2003, 2021-03-06:
@bobjacobsen  Using 4.22 (which includes PR #9422:

Examples:
I tried clicking on "Manual" and get nothing (nothing shows up in the right pane...
—> Still an issue, but now believe unrelated to the fact that manuals are PDFs.  I don’t understand why the file help/en/html/manual/index.shtml isn’t just displayed in this pane.  Doesn’t display for either TOC or Index.  The TOC and Index entries look OK and the files seems fine too.

I tried clicking through OpenLCB -> OpenLCB Signal Masts -> "technical summary on masts" (which is a PDF file) and got uninterpretable characters.
—> FIXED:  opens a browser with the linked PDF.

Thanks.

closed 2021-05-05 by stale[bot]