JMRI/JMRI #9428: ReporterOperations Jython Script Fails to load

icklesteve, 2021-02-19:
jython/ReporterOperations.py

`2021-02-18 20:25:21,548 jython.RunJythonScript                ERROR - Unable to execute script. [AWT-EventQueue-0]
javax.script.ScriptException: AttributeError: type object 'jmri.jmrit.operations.rollingstock.cars.CarManager' has no attribute 'instance' in <script> at line number 71`

silverailscolo, 2021-02-20:
You could try to add both Operations managers from lines 37 and 38 by their full path as imports.
Did the trick lately for another JMRI jython script, with a similar error.

dsand47, 2021-02-20:
The local copy of the script needs to be updated:  See PR #6418.

icklesteve, 2021-02-20:
Line number in error message may not be 100% correct!
Issue definitely in current version of the script, I couldn't see instance() within jmri.jmrit.operations.rollingstock.cars.CarManager

Note I'm not an Operations users, I was just using the script as a template for a Reporter formatting script so unable to tell if a fix is correct, just that at present it causes the error.

dsand47, 2021-02-20:
> Line number in error message may not be 100% correct!
> Issue definitely in current version of the script, I couldn't see instance() within jmri.jmrit.operations.rollingstock.cars.CarManager
>
> Note I'm [...]

My mistake,  I did not look close enough.

Try these at line 37/38:
```
    self.cm = jmri.InstanceManager.getDefault(jmri.jmrit.operations.rollingstock.cars.CarManager)
    self.em = jmri.InstanceManager.getDefault(jmri.jmrit.operations.rollingstock.engines.EngineManager)
```

The old "instance" methods are being phased out.

icklesteve, 2021-02-20:
Thanks for the correct InstancManager fix.
I then attempted to run the script and found another issue,

```
2021-02-20 14:06:45,957 jmrix.AbstractMRTrafficController     WARN  - notify: During message dispatch to MR1 [AWT-EventQueue-0]
Traceback (most recent call last):
  File "<script>", line 30, in propertyChange
  File "<script>", line 56, in format
TypeError: cannot concatenate 'str' and 'jmri.implementation.DefaultIdTag' objects

	at org.python.core.Py.TypeError(Py.java:236)
	at org.python.core.PyObject._basic_add(PyObject.java:2091)
	at org.python.core.PyObject._add(PyObject.java:2068)
	at org.python.pycode._pyx0.format$5(<script>:62)
	at org.python.pycode._pyx0.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:173)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:150)
	at org.python.core.PyFunction.__call__(PyFunction.java:426)
	at org.python.core.PyMethod.__call__(PyMethod.java:141)
	at org.python.pycode._pyx0.propertyChange$2(<script>:32)
	at org.python.pycode._pyx0.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:173)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:306)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:197)
	at org.python.core.PyFunction.__call__(PyFunction.java:485)
	at org.python.core.PyMethod.instancemethod___call__(PyMethod.java:237)
	at org.python.core.PyMethod.__call__(PyMethod.java:228)
	at org.python.core.PyMethod.__call__(PyMethod.java:218)
	at org.python.core.PyMethod.__call__(PyMethod.java:213)
	at org.python.core.PyObject._jcallexc(PyObject.java:3565)
	at org.python.core.PyObject._jcall(PyObject.java:3598)
	at org.python.proxies.__builtin__$ReporterOperations$0.propertyChange(Unknown Source)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)
	at jmri.implementation.AbstractReporter.setReport(AbstractReporter.java:61)

```

which was fixed by adding the toString().

PR #9430 (2021-03-01): Update ReporterOperations.py
Use InstanceManager, not instance()
set toString on IDTag

Fixes #9428
closed 2021-03-01 by danielb987