JMRI/JMRI #9431: Address validation / creation with Incorrect Manager

icklesteve, 2021-02-20:
Just realised it's quite easy to get the wrong manager for New Bean.

Select box choice does not always propogate.
 
eg. Inline Validation should be for Internal, not CBUS.

[image]

eg. Bean creation should be for Internal, not CBUS.

[image]

Just about to start tracking this down.

silverailscolo, 2021-02-21:
Thanks. This is point where (especially first time) users should be well served instead of getting lost, so relevant!

icklesteve, 2021-02-21:
I've got a sticking plaster fix for the inline validation, needs tidying up . . . .

danielb987, 2021-03-10:
@icklesteve Is this issue fixed? PR #9437 is merged, so if it's fixed, it would be good to close this Issue.

icklesteve, 2021-03-10:
Fixed by #9437 , although further dev ongoing Bean Table Action classes.

closed 2021-03-10 by icklesteve