JMRI/JMRI #9443: Jhelpdev has issue with files that only contain an “include”

jerryg2003, 2021-02-23:
@dsand47 found the following in working with PR #9424:

——
Buildfile: /Users/das/Git/JMRI/help/en/build.xml

init:

jhelpdev:
     [java] initial main done
     [java] start part 1
     [java] Map created from HTML/JSP files and written to ./Map.jhm
     [java] start part 2
     [java] Using temporary directory /var/folders/v3/f3h9wwb17q5gg64dtggkmq140000gp/T/.
     [java] I/O exception occurred in file './parts/SidebarInstallNew.shtml'
     [java] I/O exception occurred in file './html/tools/indexnew.shtml'
     [java] I/O exception occurred in file './parts/SidebarLayoutAutomationNew.shtml'
     [java] I/O exception occurred in file './html/hardware/SidebarUp.shtml'
     [java] I/O exception occurred in file './html/tools/tracker/Sidebar.shtml'
     [java] I/O exception occurred in file './parts/SidebarApplicationsNew.shtml'
     [java] I/O exception occurred in file './parts/SidebarToolsNew.shtml'
     [java] I/O exception occurred in file './parts/SidebarTailNew.shtml'
     [java] I/O exception occurred in file './html/tools/programmer/Sidebar.shtml'
     [java] I/O exception occurred in file './html/tools/uss/Sidebar.shtml'
     [java] I/O exception occurred in file './html/tools/fastclock/Sidebar.shtml'
     [java] I/O exception occurred in file './parts/SidebarSupportedHardwareNew.shtml'
     [java] I/O exception occurred in file './html/tools/consisttool/Sidebar.shtml'
     [java] Search database created with com.sun.java.help.search.Indexer.
     [java] start part 3

index:
     [xslt] Transforming into /Users/das/Git/JMRI/help/en
     [xslt] Processing /Users/das/Git/JMRI/help/en/JmriHelp_enIndex.xml to /Users/das/Git/JMRI/help/en/JmriHelp_enIndex.html
     [xslt] Loading stylesheet /Users/das/Git/JMRI/help/en/format.xsl
     [move] Moving 1 file to /Users/das/Git/JMRI/help/en

TOC:
     [xslt] Transforming into /Users/das/Git/JMRI/help/en
     [xslt] Processing /Users/das/Git/JMRI/help/en/JmriHelp_enTOC.xml to /Users/das/Git/JMRI/help/en/JmriHelp_enTOC.html
     [xslt] Loading stylesheet /Users/das/Git/JMRI/help/en/format.xsl
     [move] Moving 1 file to /Users/das/Git/JMRI/help/en

all:

BUILD SUCCESSFUL
Total time: 3 minutes 31 seconds
——

I have now investigated and all files giving I/O exceptions are of the same form:  they only have comments (no html), including a single “#include” of another file.  The web site builds fine with these, and the help system doesn’t use the sidebars anyway so I don’t understand why it is giving errors on those.  The only non-sidebar related file is tools/indexnew.shtml and that is a test file I was using and is not linked in anywhere (the “include” replaces the code with the contents of index.shtml).  

Is there anything I (or anyone else) should be doing further on this?

Thanks, Jerry

bobjacobsen, 2021-02-23:
JHelpDev (like JavaHelp itself) is years beyond the end-of-life point. Nobody is maintaining it.

The errors come from how it parses the content of files: It expects them to be valid HTML, and gets confused if it doesn't see the appropriate closing tags.  What message is puts out seems to depend on Java version and (perhaps) host OS.  The [JMRI JavaHelp page](https://www.jmri.org/help/en/html/doc/Technical/Help.shtml) lists a different error message.  Perhaps this one could be added to that page.

It might be possible to bypass this in _some_ cases by making the `ant` build.xml logic smarter about which files to scan, but I suspect that would be complicated to maintain.

We do have the source for JHelpDec, so somebody could perhaps fix this, but I suspect most people have higher priorities.

jerryg2003, 2021-02-23:
If ant could be made not to scan:

/help/en/parts/*
/help/en/html/tools/*/Sidebar.shtml

that would catch all but 2 of the cases and I can then go in and change those 2 files to eliminate those.

Jerry

jerryg2003, 2021-03-13:
No action necessary at this time.  REsurrect when JHelpDev discussion is reopened.

closed 2021-03-13 by jerryg2003