JMRI/JMRI #9448: Feature Request: Refactor to Create a Single, Reusable WiThrottle Client Library

mesheets, 2021-02-24:
This is an attempted clarification to a prior [feature request](https://github.com/JMRI/JMRI/issues/9446) that was quickly closed.  I'll try to explain it a different way—

Both JMRI/JMRI and JMRI/EngineDriver implement the WiThrottle protocol.  As best I have been able to tell from the source code, the WiThrottle client implementations are independent of one another.

The [WiThrottle Library Project](https://github.com/davidzuhn/WiThrottle) would demonstrate a general concept of this, but that project is written in C++ and not Java as used by JMRI and EngineDriver.  Once the WiThrottle client implementation was refactored out, it would also then be more readily available for adoption and use beyond the current use cases in JMRI/JMRI and JMRI/EngineDriver.

bobjacobsen, 2021-02-24:
I think what you're suggesting is that somebody could factor-out the WiThrottle-specific code from the existing code base, and from that create a stand-alone library that others (and also JMRI) could use.

Is that right?

JMRI's licensing allows that, so long as it's identified as coming from JMRI and all the copyrights and attributions are preserved.  It's definitely something could do.  The[ jmri.jmrit.withrottle](https://www.jmri.org/JavaDoc/doc/jmri/jmrit/withrottle/package-summary.html) package could be a starting point. I'm sure that, if somebody did that, JMRI would be happy to host it as another repository.

As you can see from the [JavaDoc](https://www.jmri.org/JavaDoc/doc/jmri/jmrit/withrottle/package-summary.html) though, that package is integrated with the rest of the JMRI code in a way that would be hard to disentangle.  It might be easier to write a WiThrottle library from scratch.

mesheets, 2021-02-24:
Yes, factor-out the WiThrottle-specific code from the existing code base, and from that create a stand-alone library that others (and also JMRI) could use.

I had looked at  [jmri.jmrit.withrottle](https://www.jmri.org/JavaDoc/doc/jmri/jmrit/withrottle/package-summary.html) and the associated [diagram and JavaDoc](https://www.jmri.org/JavaDoc/doc/jmri/jmrit/withrottle/package-summary.html) prior to writing up, and my "outsider" sense was what appears to be confirmed—that it would be more involved than extracting [jmri.jmrit.withrottle](https://www.jmri.org/JavaDoc/doc/jmri/jmrit/withrottle/package-summary.html) as-is and might be hard to disentangle from the rest of the JMRI codebase.

If starting from scratch, I would probably even try to take a step back and design the library in a somewhat more extensible manner, so that other protocols (e.g. [SRCP](http://srcpd.sourceforge.net)) could potentially be added later.  (Not that the protocols would map 1-to-1, but in the more abstract sense similar to how [JMRI supports connecting to multiple different types of command stations](https://www.jmri.org/help/en/manual/DecoderPro3/Start_Preferences.shtml).)

My availability tends to go in spurts, but I would certainly be interested in being involved with such an effort.  With a more limited scope and minimized interdependencies, it should hopefully be possible for developers to get up to speed with the project more quickly.

closed 2021-04-25 by stale[bot]