JMRI/JMRI #9450: Feature request: Multiple serial sections in a Transit alternate route

jonasbjurel, 2021-02-24:
Hi - I'm relatively new to JMRI, played around with it for a few months.
I have the following problem, which I think would be solved by allowing multiple serialized section in an alternate transit route:

I have a station yard with 5 platform tracks, 4 incoming tracks and 4 outgoing tracks. The topology is a little complex where for instance a train coming from incoming main track 1 can be switched to any of the platform tracks #1, #2, #3 and can reach outgoing main track #1, #2, #3 and #4. I want to be able to dispatch trains through the station, and automatically select the platform track based on occupancy or allocation. I can do that with alternate sections, but there is a problem with that - it will mean that the allocation happens through the entire station yard - all the way from the incoming track to the outgoing. The problem is that if a train comes in from incoming main-track #1 bound for outgoing main track #3, but should wait in any of the alternate platform tracks #1-#3, it will block a good part of the outgoing switch mesh during its entire stay at the platform - leading to bad station utilization and potential deadlocks.
What I would have wanted is to be able to have several serial sections in the alternate routes such as below
Alternate route 1: (Incoming Track #1 -> Platform 1) - (Platform 1) - (Platform 1-> Outgoing Track 3)
Alternate route 2: (Incoming Track #1 -> Platform 2) - (Platform 2) - (Platform 2-> Outgoing Track 3)
Alternate route 3: (Incoming Track #1 -> Platform 3) - (Platform 3) - (Platform 3-> Outgoing Track 3)
In this way the train waiting at the platform would only occupy the platform section during its stay, and the allocation of the exit would only happen at the time for departure.

Is my proposed approach correct, or is there other ways to achieve the same?

Best regards/
Jonas

Pugwash1, 2021-02-24:
Hi
You should use the stop allocating sensor in the transit sections. Use the same sensor in primary and alternate. You can also use a hold on the exit signal from the platforms. If the platform section includes a switch used to enter the platform, provideng its a separate block from the platform, will be released after the passes thru it.

jonasbjurel, 2021-02-25:
But this doesn't prevent that the exit is allocated during the stay in the platform, locking a lot of resources unnecessarily long, does it
BR/Jonas

jonasbjurel, 2021-03-05:
Now I see what you mean - I will try and report back - Thx.
Sorry that I'm slow - try to read-up/experiment with my relatively new experience with JMRI.
BR/Jonas

closed 2021-05-04 by stale[bot]